collabora/libsurvive #210: survive-cli failst o load driver_vive.so

PureTryOut, 2020-10-08:
**Describe the bug**
```
> survive-cli
Error loading /usr/lib/libsurvive/plugins/driver_usbmon.so: Error loading shared library driver_vive.so: No such file or directory (needed by /usr/lib/libsurvive/plugins/driver_usbmon.so)
```

Strangely enough the Alpine build system also reported `>>> ERROR: libsurvive*: driver_vive.so: path not found`. I'm not sure what it means but it seems related, I guess rpath is missing or something?

**Compile log**
https://build.alpinelinux.org/buildlogs/build-edge-aarch64/testing/libsurvive/libsurvive-0.2-r0.log

The build is of 0.2, but it's the same for the latest git master.

**Hardware setup**
Lighthouse v2 with a Valve Index.

**Desktop (please complete the following information):**
 - OS: Alpine Linux]
 - Version: edge

PureTryOut, 2020-10-08:
Also relevant:

```
bart@frostblade ~> ldd /usr/lib/libsurvive/plugins/driver_vive.so
        /lib/ld-musl-x86_64.so.1 (0x7fe6e9671000)
        libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x7fe6e9648000)
        libsurvive.so.0 => /usr/lib/libsurvive.so.0 (0x7fe6e95c5000)
        libc.musl-x86_64.so.1 => /lib/ld-musl-x86_64.so.1 (0x7fe6e9671000)
        libz.so.1 => /lib/libz.so.1 (0x7fe6e95ab000)
        libcblas.so.3 => /usr/lib/libcblas.so.3 (0x7fe6e9592000)
        liblapacke.so.3 => /usr/lib/liblapacke.so.3 (0x7fe6e9388000)
        libblas.so.3 => /usr/lib/libblas.so.3 (0x7fe6e9301000)
        liblapack.so.3 => /usr/lib/liblapack.so.3 (0x7fe6e89f0000)
        libgfortran.so.5 => /usr/lib/libgfortran.so.5 (0x7fe6e8724000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x7fe6e870a000)
        libquadmath.so.0 => /usr/lib/../lib/libquadmath.so.0 (0x7fe6e86c0000)
bart@frostblade ~> ldd /usr/lib/libsurvive/plugins/driver_usbmon.so 
        /lib/ld-musl-x86_64.so.1 (0x7f1f18152000)
        libpcap.so.1 => /usr/lib/libpcap.so.1 (0x7f1f18109000)
Error loading shared library driver_vive.so: No such file or directory (needed by /usr/lib/libsurvive/plugins/driver_usbmon.so)
        libsurvive.so.0 => /usr/lib/libsurvive.so.0 (0x7f1f18086000)
        libz.so.1 => /lib/libz.so.1 (0x7f1f1806c000)
        libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x7f1f18052000)
        libc.musl-x86_64.so.1 => /lib/ld-musl-x86_64.so.1 (0x7f1f18152000)
        libcblas.so.3 => /usr/lib/libcblas.so.3 (0x7f1f18039000)
        liblapacke.so.3 => /usr/lib/liblapacke.so.3 (0x7f1f17e2f000)
        libblas.so.3 => /usr/lib/libblas.so.3 (0x7f1f17da8000)
        liblapack.so.3 => /usr/lib/liblapack.so.3 (0x7f1f17497000)
        libgfortran.so.5 => /usr/lib/libgfortran.so.5 (0x7f1f171cb000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x7f1f171b1000)
        libquadmath.so.0 => /usr/lib/../lib/libquadmath.so.0 (0x7f1f17167000)
Error relocating /usr/lib/libsurvive/plugins/driver_usbmon.so: survive_usb_interface_str: symbol not found
Error relocating /usr/lib/libsurvive/plugins/driver_usbmon.so: survive_vive_register_driver: symbol not found
Error relocating /usr/lib/libsurvive/plugins/driver_usbmon.so: survive_dump_buffer: symbol not found
Error relocating /usr/lib/libsurvive/plugins/driver_usbmon.so: survive_data_on_setup_write: symbol not found
Error relocating /usr/lib/libsurvive/plugins/driver_usbmon.so: survive_data_cb: symbol not found
```

PureTryOut, 2020-10-09:
Actually, I mentioned this happened on 0.2 but that is not true. The build system does report the path is not found and the `ldd` output is still relevant, but running survive-cli gives me a different output:

```
> survive-cli
Info: Error.  Cannot find any valid Poser.
Info: Error.  Cannot find any valid Disambiguator.
```

With the latest git the program complains again but it does mention it loaded the HTC Vive driver.

```
> survive-cli
Error loading /usr/lib/libsurvive/plugins/driver_usbmon.so: Error loading shared library driver_vive.so.0: No such file or directory (needed by /usr/lib/libsurvive/plugins/driver_usbmon.so)
Info: Adding tracked object KN0 from HTC
Info: Adding tracked object T20 from HTC
Info: Adding tracked object KN1 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
Warning: KN0 couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
Info: T20 is treated as HMD device
Warning: KN1 couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
```

jdavidberger, 2020-10-11:
That seems like it was actually bouncing driver_usbmon but did load driver_vive. Weird RPATH stuff; I think it should be fixed in latest. 

Is there a bunch of logging messages or something in the bottom left portion of the web viewer? That is usually what congests it to that point.

PureTryOut, 2020-10-11:
Well whatever you did, the rpath stuff seems to be fixed indeed, thanks!

As for the delay:

```

Trails
FPV
Model
IMU
Shift External
95 fps

    Recording to stdout
Adding tracked object KN0 from HTC
Adding tracked object T20 from HTC
Adding tracked object KN1 from HTC
All enumerated devices attached.
Powered unit on.
Loaded drivers: HTCVive
KN0 couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
T20 is treated as HMD device
KN1 couldn't configure; probably turned off -1 LIBUSB_ERROR_IO
Detected LH gen 2 system.
LightcapMode (T20) 1 -> 2
```

Not much :shrug:

jdavidberger, 2020-10-11:
Try with `--no-record-angle` and/or '--no-record-imu`. Any help?

PureTryOut, 2020-10-12:
Yes, that helps immensely! Now the movements are recognized instantly.

jdavidberger, 2020-10-12:
Does one of the flags by itself fix the issue? I'm unsure why these overflow some systems but not other

PureTryOut, 2020-10-12:
`--no-record-angle` seems to be the one that helps. I noticed it isn't necessarily a delay, but the tracking is just really unaccurate with it enabled. When adding that flag the tracking seems to be much more accurate.

jdavidberger, 2020-10-12:
Yeah; that heavily implies that it's dropping data. A sort of unfortunate way with how the websocketd stuff in general works is that if you congest one end of that pipeline, it'll impact the CPU availability on the other end and start making USB data rates lower. It's good feedback; and I may just make `--no-record-angle` the default for `survive-websocketd`. Will have to play with that some.

closed 2020-10-12 by jdavidberger