collabora/libsurvive #212: libsurvive causes a SIGSERV in Monado

PureTryOut, 2020-10-10:
**Describe the bug**
I'm trying to get Monado running for my Valve Index, and I use libsurvive as the driver for tracking. However, the moment I turn on a controller, `monado-service` crashes and according to the stack trace the fault is somewhere in libsurvive.

**Data**
```
Reading symbols from monado-service...
Reading symbols from /usr/lib/debug//usr/bin/monado-service.debug...
(gdb) r
Starting program: /usr/bin/monado-service 
[New LWP 11483]
Error loading /usr/lib/libsurvive/plugins/driver_usbmon.so: Error loading shared library driver_vive.so.0: No such file or directory (needed by /usr/lib/libsurvive/plugins/driver_usbmon.so)
[New LWP 11484]
Info: Adding tracked object KN0 from HTC
Info: Adding tracked object T20 from HTC
Info: Adding tracked object KN1 from HTC
Info: All enumerated devices attached.
Info: Powered unit on.
Info: Loaded drivers: HTCVive
survive: Assuming variant 3
[New LWP 11485]
Waiting for survive HMD config parsing
Waiting for survive HMD config parsing
Waiting for survive HMD config parsing
Waiting for survive HMD config parsing
Waiting for survive HMD config parsing

Thread 4 "monado-service" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 11485]
survive_attach_config (ctx=ctx@entry=0x7ffff35e2150, tag=tag@entry=0x7ffff4d2c5d6 "max-error", 
    var=var@entry=0x7ffff4b1bc48, type=type@entry=102 'f')
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_config.c:1017
1017    /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_config.c: No such file or directory.
(gdb) bt
#0  survive_attach_config (ctx=ctx@entry=0x7ffff35e2150, tag=tag@entry=0x7ffff4d2c5d6 "max-error", 
    var=var@entry=0x7ffff4b1bc48, type=type@entry=102 'f')
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_config.c:1017
#1  0x00007ffff4ccf9b6 in survive_attach_configf (ctx=ctx@entry=0x7ffff35e2150, 
    tag=tag@entry=0x7ffff4d2c5d6 "max-error", var=var@entry=0x7ffff4b1bc48)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_config.c:1080
#2  0x00007ffff4cc95d7 in general_optimizer_data_init (d=d@entry=0x7ffff4b1bc30, so=so@entry=0x7ffff35a5220)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/poser_general_optimizer.c:29
#3  0x00007ffff3629f76 in PoserMPFIT (so=0x7ffff35a5220, user=0x7ffff35a56e8, pd=0x7ffff357dca0)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/poser_mpfit.c:836
#4  0x00007ffff4cdd983 in survive_default_imu_process (so=0x7ffff35a5220, mask=3, accelgyromag=0x7ffff357e170, 
    timecode=2618740736, id=0)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_process.c:194
#5  0x00007ffff4cdd7ae in survive_default_raw_imu_process (so=0x7ffff35a5220, mask=<optimized out>, 
    accelgyromag=<optimized out>, timecode=2618740736, id=0)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_process.c:172
#6  0x00007ffff3601e60 in read_imu_data (w=w@entry=0x7ffff35a5220, time=time@entry=39958, 
    readPtr=readPtr@entry=0x7ffff357e6c8, 
    payloadEndPtr=payloadEndPtr@entry=0x7ffff35b4a12 "\233\rM\020 P\223\004\236\025\277\003\213\375N\345\324\r[p\372\002@\352N\217\213\343L")
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:1769
#7  0x00007ffff3604dcb in handle_watchman_v2 (
    payloadEndPtr=0x7ffff35b4a12 "\233\rM\020 P\223\004\236\025\277\003\213\375N\345\324\r[p\372\002@\352N\217\213\343L", payloadPtr=0x7ffff35b49f8 "\220Ў\373\225\361\025\372\377\377\005", time=<optimized out>, w=0x7ffff35a5220)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:2283
#8  handle_watchman (w=0x7ffff35a5220, readdata=<optimized out>)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:2413
#9  0x00007ffff360649a in survive_data_cb_locked (si=si@entry=0x7ffff35b49c8)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:2841
#10 0x00007ffff3606cd8 in survive_data_cb (si=0x7ffff35b49c8)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:564
#11 0x00007ffff36007df in handle_transfer (transfer=0x7ffff4597f00)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.libusb.h:175
#12 0x00007ffff7eb75e6 in ?? () from /usr/lib/libusb-1.0.so.0
#13 0x00007ffff7eba5fd in ?? () from /usr/lib/libusb-1.0.so.0
#14 0x00007ffff7ebcb01 in ?? () from /usr/lib/libusb-1.0.so.0
#15 0x00007ffff7eb7482 in ?? () from /usr/lib/libusb-1.0.so.0
#16 0x00007ffff7eb7cee in libusb_handle_events_timeout_completed () from /usr/lib/libusb-1.0.so.0
#17 0x00007ffff7eb7e3c in libusb_handle_events_completed () from /usr/lib/libusb-1.0.so.0
#18 0x00007ffff7eb8613 in ?? () from /usr/lib/libusb-1.0.so.0
#19 0x00007ffff7eb88ee in libusb_control_transfer () from /usr/lib/libusb-1.0.so.0
#20 0x00007ffff3601f24 in getupdate_feature_report (datalen=256, data=0x7ffff3583bb8 "\021\252\024", interface=0, 
    dev=0x7ffff435cf20)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5--Type <RET> for more, q to quit, c to continue without paging--
296/src/driver_vive.libusb.h:52
#21 hid_get_feature_report_timeout (len=256, buf=0x7ffff3583bb8 "\021\252\024", iface=0, device=0x7ffff435cf20)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:634
#22 get_feature_report_timeout_locked (ctx=ctx@entry=0x7ffff35e2150, device=device@entry=0x7ffff435cf20, 
    buf=buf@entry=0x7ffff3583bb8 "\021\252\024", len=len@entry=256, iface=0)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:646
#23 0x00007ffff3602184 in survive_get_config (config=config@entry=0x7ffff3597d70, 
    usbInfo=usbInfo@entry=0x7ffff35b4ea8, send_extra_magic=<optimized out>, iface=0, sv=<optimized out>, 
    sv=<optimized out>)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:971
#24 0x00007ffff36028c4 in LoadConfig (iface=0, usbInfo=0x7ffff35b4ea8, sv=0x7ffff35b4990)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:3107
#25 survive_vive_usb_poll (ctx=0x7ffff35e2150, v=0x7ffff35b4990)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/driver_vive.c:853
#26 0x00007ffff4ccc1ee in survive_poll (ctx=0x7ffff35e2150)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive.c:905
#27 0x00007ffff4ccc8d8 in __simple_thread (_actx=0x7ffff3641130)
    at /home/bart/Documents/Git/alpine/aports/testing/libsurvive/src/libsurvive-526e06133f837e1cd526bd6bc6eab0c88d8d5296/src/survive_api.c:327
#28 0x00007ffff7fbaaae in ?? () from /lib/ld-musl-x86_64.so.1
#29 0x0000000000000000 in ?? ()
```

**Hardware setup**
Lighthouse v2 with a Valve Index.

**Desktop (please complete the following information):**
 - OS: Alpine Linux
 - Version edge

ChristophHaag, 2020-10-10:
I experimented a bit with alpine edge. survive-cli runs, but survive_api crashes.

When compiling libsurvive with clang, api_example crashes immediately in a libsurvive function where the function call is somehow corrupted
`CC=clang CXX=clang++ cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_INSTALL_PREFIX=/usr -G Ninja -B build`
https://gist.github.com/ChristophHaag/bf47dc32d6a66a28559010db86c76b05

When compiling libsurvive with gcc, api_example runs for a short while and then crashes in lapack
`cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_INSTALL_PREFIX=/usr -G Ninja -B build`
https://gist.github.com/ChristophHaag/4a928e3aca4ac36197ca935b7e52f604

The only clue is "Bad permissions for mapped region" from valgrind
```
==3817476== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==3817476==  Bad permissions for mapped region at address 0x5D29C78
==3817476==    at 0x4CBA6DA: general_optimizer_data_init (poser_general_optimizer.c:23)
==3817476==
==3817476== Process terminating with default action of signal 11 (SIGSEGV)
==3817476==  Bad permissions for mapped region at address 0x5D29C30
==3817476==    at 0x489813F: _vgnU_freeres (vg_preloaded.c:81)
```
Guessing it's something with musl...

jdavidberger, 2020-10-11:
@ChristophHaag That crash is sorta wild; `general_optimizer_data_init` is saying it got a null pointer for `so` but `PoserMPFIT` didn't -- even though it's a straight pass through. 

@PureTryOut That crash is also weird; don't see how it would even consider segfaulting at that point in the code. 

It almost looks like a calling convention mishap; but usually those are across binaries which isn't the case here. Also possibly something is just smashing the stack but I'm not sure what. 

I fixed a bug address sanitizer uncovered but I doubt that was the issue. 

I have it running in a virtual machine and I couldn't get api_example to crash; gcc or otherwise. Do either of you have core dumps or anything to go with?

PureTryOut, 2020-10-11:
> Do either of you have core dumps or anything to go with?

Not currently. How would I get one?

jdavidberger, 2020-10-11:
`
echo '/tmp/core.%h.%e.%t' > /proc/sys/kernel/core_pattern
ulimit -c unlimited
`

All this is on x64? Are you running on a VM or anything?

ChristophHaag, 2020-10-11:
Running `generate-core-file` in gdb also works.

Personally I run in a chroot with /proc, /sys, and /dev bind mounts, on 64 bit x86.

Here's my build directory including a `core.137829`, should work with `gdb ./api_example core.137829`
compiled with flags
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_INSTALL_PREFIX=/usr -G Ninja -B build
[build.tar.gz](https://github.com/cntools/libsurvive/files/5362289/build.tar.gz)

I can confirm that it runs fine with only the Index HMD. I get this crash when at least one Index Controller is also powered on (rf connected).

PureTryOut, 2020-10-12:
Yes on x86_64 (amd64), I run it on bare metal, Alpine Linux is my daily distribution of choice.

I build it as the following:

```
cmake -B build \
    -DCMAKE_BUILD_TYPE=None \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_INSTALL_LIBDIR=lib \
    -DBUILD_STATIC=OFF
```

[core.8993.tar.gz](https://github.com/cntools/libsurvive/files/5363441/core.8993.tar.gz)

jdavidberger, 2020-10-12:
If you add `-Wl,-z,stack-size=2000000` to the flags, it'll fix it.

What is happening is that it's overflowing the stack space. Musl wants to give 128k; glibc usually gives a few MB. Libsurvive aggressively uses stack allocation when it can to avoid hitting the heap for anything -- which works well assuming you don't have a stack space smaller than you are going to use. 

The reason the stack traces look weird is that it's crashing when first trying to put something in the overflowed stack region; since those arguments are out of valid memory they show up as 0x0. 0xffffffffff600000 is a red herring -- that is syscallv space that GDB is just trying to collect for no real good reason. 

I think I'll look at adding a compilation flag or something that sorts that out intelligently. For now you might have to just patch that linker flag in.

jdavidberger, 2020-10-12:
This is fixed in latest commit. When we create threads we now put their max size at 8mb; which is excessive but anything using the pthread stuff has virtual memory and won't really notice. Can either/both of you confirm the fix?

PureTryOut, 2020-10-12:
Ah yes stack size, I recently had that same problem with a different project. The issue is fixed now, thanks! I can now view and control OpenXR applications using Monado and libsurvive :smile:

closed 2020-10-12 by PureTryOut

jdavidberger, 2020-10-12:
Perfect! Yeah; the dead giveaway for me was the instruction it segfaulted on was just saving some bit of memory to the stack frame; which I hadn't quite expected. 128kb is sort of insanely small for a linux system nowadays; and it's sort of annoying -- the reason you might have seen this with the `api-example` and not `survive-cli` or in the unit tests is that the main stack was probably allocated ~2-8mb and could run that bit of code without issue. 

To be honest; it's a reminder at how primitive some of the C/C++ stack is. A lot of things could take that segfault and check it against that threads stack size; but there isn't a real clean way to do it so it just says 'SEGFAULT' and if you've never seen it before you'd be at a loss for a while over it.