collabora/libsurvive #216: isnan(out[j] -- survive_optimizer.c

joejarvis64, 2020-11-24:
**Describe the bug**
I keep running into a random issue where I come down with an isnan exception.

Interestingly, everything seems to be tracking fine for a while.  However, I only crash in this bit of code when I start to raise the HMD over my head.  

I am running the api_example.c program with no command line options.
I am running on windows have 3 gen2 lighthouses setup and a valve index headset.

Any ideas on how to resolve this issue?

SimpleApi: Adding tracked object KN0 from HTC
SimpleApi: Adding tracked object T20 from HTC
SimpleApi: Adding tracked object KN1 from HTC
SimpleApi: All enumerated devices attached.
SimpleApi: Could not turn on KN0(2) (-1/19 - )
SimpleApi: Could not turn on KN0(2) (-1/9 - )
SimpleApi: Could not turn on KN0(2) (-1/6 - )
SimpleApi: Could not turn on KN1(2) (-1/19 - )
SimpleApi: Could not turn on KN1(2) (-1/9 - )
SimpleApi: Could not turn on KN1(2) (-1/6 - )
SimpleApi: Powered unit on.
SimpleApi: Loaded drivers: HTCVive
Found 'KN0'
Found 'T20'
Found 'KN1'
Found 'LH0'
Found 'LH1'
Found 'LH2'
LH0 LHB-7FFFFFFF (0): -2.201303 -0.387905 1.557917 -0.625820 -0.530292 0.384218 0.423694
SimpleApi: KN0 couldn't configure; probably turned off -1
SimpleApi: T20 is treated as HMD device
SimpleApi: KN1 couldn't configure; probably turned off -1
SimpleApi: Detected LH gen 2 system.
SimpleApi: LightcapMode (T20) 1 -> 2
SimpleApi: Position for LH0 seems bad; queuing for recal
SimpleApi: Position for LH1 seems bad; queuing for recal
SimpleApi: Too many failures for T20; reseting calibration 3.652086e-01 ( 0.1028 stationary)
SimpleApi: LH0 0.156943
SimpleApi: LH1 0.238092
SimpleApi: LH2 0.000000
SimpleApi: Attempting to solve for 0 with 29 meas from device T20
SimpleApi: Attempting to solve for 1 with 28 meas from device T20
SimpleApi: Attempting to solve for 2 with 32 meas from device T20
SimpleApi: Assuming object position of +0.000000e+00   +0.000000e+00   +0.000000e+00    +1.000000e+00   +0.000000e+00   +0.000000e+00   +0.000000e+00
SimpleApi: Solved for 0 with error of 0.072102/0.0000017711 (acc err 1.0249)
SimpleApi: Solved for 1 with error of 0.072102/0.0000017711 (acc err 0.9759)
SimpleApi: Solved for 2 with error of 0.072102/0.0000017711 (acc err 0.9888)
SimpleApi: Using LH 0 (7fffffff) as reference lighthouse
T20 LHR-65CA2B2E (9): -0.046371 -0.072606 -0.037248 -0.704690 -0.707858 -0.032226 -0.036199
T20 LHR-65CA2B2E (9): -0.045998 -0.072555 -0.035731 -0.704261 -0.708308 -0.031087 -0.036735
T20 LHR-65CA2B2E (9): -0.045871 -0.072589 -0.035066 -0.703571 -0.709032 -0.032068 -0.035118
T20 LHR-65CA2B2E (10): -0.045932 -0.072539 -0.034578 -0.703941 -0.708659 -0.030989 -0.036195
T20 LHR-65CA2B2E (10): -0.045901 -0.072611 -0.034554 -0.703660 -0.708938 -0.031184 -0.036019

...

T20 LHR-65CA2B2E (23): -1.103515 0.489729 -0.323331 -0.921828 -0.315982 -0.102321 -0.199794
T20 LHR-65CA2B2E (23): -1.089329 0.499535 -0.303968 -0.922072 -0.349686 -0.113786 -0.120647
T20 LHR-65CA2B2E (23): -1.071633 0.514748 -0.299548 -0.894818 -0.393376 -0.048541 -0.205427
T20 LHR-65CA2B2E (24): -1.074699 0.490735 -0.304079 -0.924625 -0.315381 -0.059594 -0.205067
T20 LHR-65CA2B2E (24): -1.067517 0.419435 -0.305101 -0.935818 -0.193334 -0.115393 -0.271202
T20 LHR-65CA2B2E (24): -1.054296 0.421147 -0.320278 -0.953968 -0.235215 -0.076406 -0.169651
T20 LHR-65CA2B2E (24): -1.033854 0.391038 -0.307386 -0.986546 -0.159906 0.032317 0.010642
T20 LHR-65CA2B2E (25): -1.018904 0.341667 -0.313803 -0.996638 -0.047539 0.056422 0.035630
T20 LHR-65CA2B2E (25): -1.012918 0.394011 -0.285817 -0.979948 -0.089848 0.134008 -0.116926
T20 LHR-65CA2B2E (25): -1.095919 0.534112 -0.048963 -0.987878 -0.130339 0.051228 -0.066970
T20 LHR-65CA2B2E (25): -1.165657 0.660977 0.581114 -0.827321 -0.552357 -0.094024 -0.040020
Assertion failed: !isnan(out[j]), file C:\develop\libsurvive\src\survive_optimizer.c, line 281

jdavidberger, 2020-11-25:
If you build it in release mode, it should make the asserts not fire. I think the process should be pretty resilient to NaNs there; but I've never seen one there that wasn't a bug from something else so I'm curious if release mode tracks correctly for you.

joejarvis64, 2020-11-25:
This is kind of an interesting issue.  I have some repeatable steps to re-create this particular error.  I arranged my lighthouses in a triangle.  They are each about 6-7 feet apart.  two of the lighthouses are 6 feet up.  the third is desk level about 3 feet up.

If i take the hmd over near the lighthouse at desk level, and then raise it to about 6 feet, it will crash.  

I was able to re-create that several times.  Another interesting note is that I turned on a fourth lighthouse and it would not see that issue.  Seems pretty odd, but that scenario caused a crash for me multiple times.

I just saw your post so will try again in release mode.

joejarvis64, 2020-11-25:
While in release mode, it did not crash, but the app stopped sending poses to the screen.  Guessing it took an exception and something else stopped.  Let me know if there is any other debug info I could gather to help figure out the issue.  I ran with -v 150 and this was the end of the logging to the screen.  

SimpleApi: MPFIT success T20 8.0420907.5s 0.009177/0.0040250783 (64 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0603587.5s 0.026377/0.0032804395 (46 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0614377.5s 0.017677/0.0042727576 (46 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0620387.5s 0.017658/0.0047156486 (53 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0794637.5s 0.017785/0.0023234060 (47 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0801357.5s 0.013987/0.0044396598 (49 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0807117.5s 0.007412/0.0019359197 (48 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0986137.5s 0.023152/0.0035954354 (57 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.0998087.5s 0.014450/0.0035144664 (57 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1010487.5s 0.005146/0.0014714147 (57 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1164177.5s 0.014841/0.0029572017 (60 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1169107.5s 0.014762/0.0032168608 (60 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1194067.5s 0.017199/0.0046714780 (60 measurements, 2 result, 3 lighthouses)
T20 LHR-65CA2B2E (11): -0.076648 -1.465379 0.778628 0.247042 0.372477 0.708358 0.546315
SimpleApi: MPFIT success T20 8.1375057.5s 0.013531/0.0029747974 (56 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1382437.5s 0.013416/0.0030449433 (57 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1404987.5s 0.008048/0.0018417022 (57 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1544817.5s 0.020319/0.0031460076 (55 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1554847.5s 0.020190/0.0034105027 (55 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1594477.5s 0.008173/0.0022771585 (50 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1734647.5s 0.008356/0.0011678611 (49 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1754537.5s 0.013874/0.0012649940 (49 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1797057.5s 0.010073/0.0021671510 (49 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1925117.5s 0.004613/0.0012581667 (51 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1948107.5s 0.004545/0.0011856535 (51 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.1993917.5s 0.003873/0.0014529869 (52 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.2104037.5s 0.007309/0.0011454426 (50 measurements, 2 result, 3 lighthouses)
SimpleApi: MPFIT success T20 8.2183757.5s 0.005976/0.0005097012 (46 measurements, 2 result, 3 lighthouses)

jdavidberger, 2020-11-25:
This is the kind of data log I'd need to look at it: https://github.com/cntools/libsurvive#normal-recording

Sounds like it's confused on a lighthouse position or something; it looks like it has plenty of actual data though so I'm not sure what is going on.

joejarvis64, 2020-11-25:
I attached the recording.  Another interesting item is that the config.json ended up changing the id of all the lighthouses to 0x7fffffff.  Prior to the start of the run they were set to the proper serial number.

[nan_crash.rec.gz](https://github.com/cntools/libsurvive/files/5597076/nan_crash.rec.gz)

jdavidberger, 2020-11-25:
The tracking for me from that recording seems to not fall off and seems right: 

[image]

The lighthouse ID thing is a bug that I need to drill into and fix; but I don't think it's what you are seeing here. 

I didn't see any NaNs though. There are some differences in the math calls between windows and linux. I'm going to fire this up on windows and see if it's one of those differences that is causing you to see these while mine doesn't.

joejarvis64, 2020-11-25:
that recording abruptly stopped.  I was wondering if perhaps a buffer did not get flushed so maybe was missing a few lines at the end?

jdavidberger, 2020-11-25:
Oh interesting. Possibly; see if you can run it without any posers at all by giving it `--poser Dummy`. Then just take a long recording(>25seconds) doing the same basic motions. If it's crashing the entire thing then I could see it not making it to the log file but I am surprised it impacts it that much.

joejarvis64, 2020-11-25:
I recorded with the dummy poser and it did not crash with that poser as I did the same movements.  I then repeated the tests without the dummy poser just to make sure i could still crash it again today and it did crash as expected.  I uploaded the recording if you want to take a peek.

[nan_crash_dummy_poser.rec.gz](https://github.com/cntools/libsurvive/files/5599068/nan_crash_dummy_poser.rec.gz)

jdavidberger, 2020-11-25:
Weird; so the thing that was nan/inf is caused by the HMD being kind of near the boundary of what it thinks the LH boundary is. It thinks its slightly outside the boundary, but has a measurement from it which screws the math up. It doesn't seem to happen on linux with this recording for whatever reason -- floats are chaotic and weird and it's not the most deterministic algo in that respect. 

Anyway; I pushed a fix to master that should address it. Mind if I add your rec files to the unit tests?

joejarvis64, 2020-11-25:
Awesome! I tried out this new version and it is working as expected now.  

Thanks for fixing this up so quickly.  Feel free to add those rec files to the test suite.

closed 2020-12-11 by joejarvis64