collabora/libsurvive #217: Unknown model_number 'VIVE_Pro MV'

dikonov, 2020-11-27:
While running OpenXR runtime monado-service, compiled with ibsurvive, I get the message. `Unknown model_number 'VIVE_Pro MV'. Please submit an issue`.  

Here is the full log:
```
Info: Device T20 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x85020006
Warning: The detected version for device T20 is 1526689969; the earliest that is verified to work is 1541806442. You may want to upgrade. If this version seems to work, please create an issue at https://github.com/cntools/libsurvive/issues with this message so we can update the version list.
Warning: Unknown model_number 'VIVE_Pro MV'. Please submit an issue with this value describing your device so it can be added to the known list.
Info: T20 is treated as HMD device
survive got HMD config
Waiting for survive HMD to be found...
Found HMD: T20
survive HMD present
Survive eye resolution 1440x1600
DEBUG [init_valve_index] Valve Index manufacturer string did not match.
Could not parse JSON data.Could not parse JSON data.Skipping libsurvive initialization, already initialized
open_hid_interface - Failed to open device '/dev/hidraw4' got '-2'
ERROR [init_vive_pro] Could not open Vive sensors device.
open_hid_interface - Failed to open device '/dev/hidraw9' got '-2'
ERROR [vive_controller_found] Could not open Vive controller device.
open_hid_interface - Failed to open device '/dev/hidraw10' got '-2'
ERROR [vive_controller_found] Could not open Vive controller device.
DEBUG [xrt_gfx_provider_create_native] Doing init 0x1d9ad20
WARN [compositor_check_and_prepare_xdev] Had to fill in meshuv on xdev 'Survive HMD', this should be done in the driver.
INFO [comp_resources_init] New renderer initialized!
Created listening socket.
Listening socket is fd 51
```
The lighthouses at the time of running are OFF (temporarily not available).

**Setup details**
HTC VivePro connected to an eGPU with RTX2070 card inside. The OS is Linux 5.4.74.

The HMD  seems to work.  I am able to run SteamVR. I can either switch to the Extended mode and see the non-composited mono desktop picture or use the direct mode and see nothing. 

**Additional context**
I want to run OpenMW VR fork, but I see nothing in the HMD. All output ends up in a window on the regular display. 
I cannot get the proper composited VR picture in the HMD. The game itself is nicely controlled by keyboard/mouse and might be playable even without lighthouses and VR controllers.

I wonder if this problem is due to the cause of the warning.

jdavidberger, 2020-11-27:
Thanks for filing the report! Is there a new version of firmware available in steamvr? This is saying it's a little older than what I thought was the most recent. 

The log is good and I'm going to add that model string into the list of known ones so it maps out correctly; but I think it won't solve your ultimate problem. I think that without lighthouses turned on or available, it simply will never output a position for the HMD, and I imagine it's waiting on at least one position before it displays anything. I'm not sure how it'd know what to display without that; it could place the HMD at origin but origin is usually on the floor and would be not ideal.

dikonov, 2020-11-28:
Refusing to display video without controllers is stupid. I can run Safespaces VR desktop in an SBS stereo window, and it receives gyro data and rotates correctly. The user is always in the center of the virtual space and never moves anywhere. As for the OpenMW game, it does not actually need any 3d controllers to be playable. It is possible to walk around and interact using keyboard and mouse.  

I am not sure that updating firmware is possible in SteamVR and I do not know how to find it. I do not use Windows, and there are known problems with flashing fw in Linux. Maybe, I'll find someone else to reflash it.

jdavidberger, 2020-11-28:
In steamvr the device icons will have an indicator for firmware updates. If you don't see that, it's likely up to date. 

Is it the controllers that are off or the lighthouses? Is expect it to work without controllers. But since the lighthouse boxes are what let it know where it is in space, it seems reasonable to me not to just render a random location.

dikonov, 2020-11-28:
I do not have the lighthouses right now, only HMD and controllers. I need to know, if an incomplete setup with just HMD can be used to watch 3D video, and generally see any VR picture without moving in VR. SteamVR cannot do that. 

There is no fw update icon in SteamVR for this HMD.

jdavidberger, 2020-11-28:
If you pass "--poser kalmanonly" through to libsurvive it might sort of do what you want. That's a mode typically used for debugging things so I'm not sure how well it works with an hmd.

jdavidberger, 2020-12-21:
This issue should be fixed in 234e881827b4d557c21171f7cd1ab9aa487bf9d2

closed 2020-12-21 by jdavidberger