collabora/libsurvive #219: Support for Pimax 8K X

paju1986, 2020-12-21:
The pimax 8k X use lighthouse tracking , would it be possible to add support for it?

jdavidberger, 2020-12-21:
I don't have a pimax to test with; but based on how the lighthouse ecosystem typically works there is a decent chance just plugging it in might just work; or at least provide specific things that need to be addressed -- such as understanding the model number or FW revision. 

If it doesn't enumerate at all, either paste the output of (from linux) `lsusb` or grab a hardware log from steamvr in either windows or linux and post it here. It might just be changed USB descriptors.

ChristophHaag, 2021-04-27:
I have a Pimax 5k plus now and it tracks "out of the box" with libsurvive, didn't even need to add a usb id.

For me the display doesn't turn on, we might need to send some usb commands to enable it. There are PCAP captures presumably using the official pimax driver in this issue https://github.com/OpenHMD/OpenHMD/issues/317

thaytan speculated on = `0200002001e803`, off = `0200002000e803` for the 5k and on = `02000000010000`, off = `02000000000000` for the 8k

jdavidberger, 2021-12-19:
Revisiting old issues; and I'm not sure the remedy here necessarily. I think driving the HMD visuals falls outside the realm of libsurvive; but I'm not sure the best way forward when a device needs to be configured over USB. If anyone at openhmd wants to coordinate in a way that I can give access to the libusb subsystem I think that would work but they'd likely have to have a pimax and want to suss out the protocol; since I don't have one.

SpookySkeletons, 2023-10-02:
Posting for anyone curious, it looks like it will work for all P2 HMDs
It's a dual keepalive packet loop.
https://gitlab.freedesktop.org/haagch/monado/-/commits/5k+

SpookySkeletons, 2024-08-07:
Update:

We have full pimax support for vive, survive, and steamvr_lh drivers in monado as well as vive pro 2 with this branch https://gitlab.freedesktop.org/Coreforge/monado/-/tree/pimax

You still must patch your kernel and bodge the EDID a bit in kernel parameter loader to make the kernel not sick upon seeing them.
Pimax also needs distortion mesh dumps from pitools SDK.