collabora/libsurvive #223: Sharp power consumption surges

dikonov, 2021-01-16:
My computer halts and powers off immediately as soon as I run `survive-websocketd` + `websocketd`.  The same system successfully passed calibration process in SteamVR several times and I can run SteamVR and some VR apps reliably. Some other apps halt the system too. I am pretty sure that the issue boils down to **power efficiency of the software**.

I can run `survive-cli` without this issue, but I have no idea how to complete libsurvive calibration using the CLI version. 

**Important context**
I have a capable z490 mainboard + intel 10700 CPU packed into a tiny mini-ITX format case, which requires an external 12V DC adapter. Unfortunately, the adapter is still somewhere in the delivery chain and I have to use an underpowered temporary one.  The temporary adapter provides up to 120W, which is way too little, and makes the system extremely sensitive to CPU load and CPU power consumption. (*My GPU is external and has its own independent power supply).

_Perhaps, it is worth doing something to lower the CPU load and conserve power even for normally powered systems._ SteamVR demonstrates better efficiency in this domain.

**Data**
No logs or errors are saved during sudden hardware poweroffs.

**Hardware setup**
intel 108700 CPU (65W TDP)
MSI z490 Unfy mainboard
RTX2070 eGPU (independently powered)
HTC VivePro HMD + two LH 2.0 + HTC Vive controllers.

**Desktop (please complete the following information):**
 - OS: Linux (ALTLinux Sisyphus)
 - Version:
          Linux kernel 5.4.74,
          libsurvive git from 2012.01.16,
          nvidia blob 450.80.02

jdavidberger, 2021-01-16:
I'm guessing it's the 3d rendering and/or the communications pipeline that is an issue here. Even with an externally powered GPU transferring buffers back and forth can be a bit of drain. 

Calibrating via `survive-cli` is pretty much the exact same as with the websocketd interface; just you can't immediately see the output. `survive-cli --force-calibrate` will force a recalibration even if it already has a good one. 

Other than that, you can do a few things to lower power requirements: 
- Make sure your browser is using GPU hardware -- https://browserleaks.com/webgl
- Run with `--no-record-rawlight --no-record-imu --no-record-angle`. This will cut down on the output and might be enough to make a difference.

dikonov, 2021-01-17:
Thanks! It helped. 
I can see the HMD and controllers moving in the browser window now. The tracking stops after several minutes. Does it mean that the calibration has completed?

jdavidberger, 2021-12-19:
This was probably a crash; but it was a while ago so I think this was resolved in discord.

closed 2021-12-19 by jdavidberger

dikonov, 2021-12-24:
WTF is discord? The bug is in git issues and is dicussed here. This is a power efficiency issue. It can be resolved by optimizing the code or circumvented by providing more electric power and hurting the climate in process. 

Just closing a bug without doing anything is weird. 
Chatting about it wothout doing anything is pointless and weird.

jdavidberger, 2021-12-24:
I thought we might have talked on the discord chat server about it; but apparently not. I already responded to your power efficiency question. I'm open to any optimizations you want to suggest but again, I'm pretty sure the power spike was from your GPU which no amount of optimization in this library can address.

dikonov, 2021-12-27:
I am sure that GPU was not the main issue here, because this machine uses an external detachable eGPU in a separate box with its own autonomous power supply (which covers 100% of the GPU power consumption).  I admit that the box is electrically underpowered for VR with just 150W picoPSU (+ 250W dedicated GPU power), but ITX form factor sets physical limits. I even had to replace a molten power jack...   On the other hand it is is a good test bench for such problems.