collabora/libsurvive #224: Warnings while running monado-service with a VivePro

dikonov, 2021-01-17:
**Describe the bug**
I am finally able to test a complete VR setup. It seems to work, but I get many cryptic warnings and suggestions to create an issue about it:

1. These messages appear in the output of monado-service: 
```
 Warning: The detected version for device T20 is 1526689969; the earliest that is verified to work is 1541806442. You may want to upgrade. If this version seems to work, please create an issue at https://github.com/cntools/libsurvive/issues with this message so we can update the version list.

Info: Device WM1 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x81020006
Warning: The detected version for device WM1 is 1526689969; the earliest that is verified to work is 1541806442. You may want to upgrade. If this version seems to work, please create an issue at https://github.com/cntools/libsurvive/issues with this message so we can update the version list.

Info: Device WM0 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x81020006
Warning: The detected version for device WM0 is 1526689969; the earliest that is verified to work is 1541806442. You may want to upgrade. If this version seems to work, please create an issue at https://github.com/cntools/libsurvive/issues with this message so we can update the version list.

```
**SteamVR tells me that no firmware updates are available for the HMD and controllers**. There is some update for the lighthouses, but I am afraid to install it because of many reports that updating lighthouses does not work in Linux and can even brick them.  

2. I am able to run calibration via survive-websocket and see the VR devices being tracked now. However, running monado-service results in an endless stream of messages like this:
```
Warning: Not entirely sure what this data is; errors may occur (0, 0x53)

53 4a   03 01 
Warning: Not entirely sure what this data is; errors may occur (0, 0xa3)

a3 70   85 01 
Warning: Not entirely sure what this data is; errors may occur (0, 0xe3)

e3 96   07 02 
Warning: Not entirely sure what this data is; errors may occur (0, 0x33)

33 bd   89 02 
Warning: Not entirely sure what this data is; errors may occur (0, 0x83)

83 e3   0b 03 01 10   85 ab 00 56   9b 7c 21 46   a0 7c 09 8e   b3 7c 01 
Warning: Not entirely sure what this data is; errors may occur (0, 0xc3)

c3 09   8e 03 
Warning: Not entirely sure what this data is; errors may occur (21, 0x13)

01 50   0d e6 04 02   ac f1 21 ba   b0 f1 09 42   c4 f1 01 b6   cf f1 a9 13   30 10 04 
Warning: Not entirely sure what this data is; errors may occur (0, 0x43)

```
I attempted to run the VR (OpenXR based) fork of the OpenMW (https://gitlab.com/madsbuvi/openmw/-/tree/openxr_vr), but the game does not detect the presence of working VR gear and continues to run in 2D. The HMD goes active but shows only light grey and no picture. 

**Hardware setup**
HTC VivePro HMD + two LH 2.0 + HTC Vive controllers.
intel 108700 CPU (65W TDP)
MSI z490 Unfy mainboard
RTX2070 eGPU

**Desktop (please complete the following information):**
OS: Linux (ALTLinux Sisyphus)
Version:
Linux kernel 5.4.74,
libsurvive git from 2012.01.16 (includes the fix from https://github.com/cntools/libsurvive/issues/217),
nvidia blob 450.80.02

**Additional context**
The devices work OK under SteamVR.

dikonov, 2021-01-17:
Monado-service before calibration
[monado-service-complaints.txt](https://github.com/cntools/libsurvive/files/5827044/monado-service-complaints.txt)

Output of survive-cli during calibration
[monado-service-complaints.txt](https://github.com/cntools/libsurvive/files/5827053/monado-service-complaints.txt)

Monado-service running after calibration (part)
[survive-cli-goodrun.txt](https://github.com/cntools/libsurvive/files/5827050/survive-cli-goodrun.txt)

OmarZahra, 2023-05-01:
Hi, I am having the same issue with vive pro2. Can you share how you solved the issue please?