collabora/libsurvive #228: How many base stations (Lk

garbage98, 2021-02-09:
**Describe the bug**
A clear and concise description of what the bug is.

**Data**
For tracking or device specific issues, it is incredibly helpful if you could provide us with a data log of the issue. See the [readme](https://github.com/cntools/libsurvive#data-recording) for how to do that.

**Compile log**
If the issue is a compilation issue, please provide a log of the compiler output; as well as what compiler you are using. 
**Hardware setup**
Please include the hardware setup, such as what devices you are using and which version of lighthouse you are using (The rounded face base stations are version 2, the flat face base stations are version 1). 

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, Mac, Linux]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

JohanVer, 2021-08-06:
I am also curious about this. Does anybody know?

BenWoodford, 2021-08-24:
16 as there’s no special handling for overlapping channels right now, IIRC?

jdavidberger, 2021-12-19:
It is 16 currently

closed 2021-12-19 by jdavidberger