collabora/libsurvive #235: Support for VIVE Tracker 3.0

CarlosDanielFP, 2021-05-19:
Hi! I was wondering if there is already any effort in adding support for the latest VIVE trackers (v3.0). I am working on a project and the team bought these (Unity would be a waste of resources).
I have never done anything like this but I can try to contribute to libsurvive by adding said support. I just need some guidance for that to happen 😅 please! 🥺

jdavidberger, 2021-05-19:
Tracker 3.0 should just work right now; I have one and have used it. Are you seeing any issues with it?

CarlosDanielFP, 2021-05-19:
I am trying to use it with a Raspberry Pi 4B. I'm getting these messages:
```
Warning: Too many failures for WM0; reseting calibration 2.298801e-01 ( 0.3535 stationary)
Warning: WM0 is probably dropping IMU packets; 0.539701 time reported between 3370990848
```

CarlosDanielFP, 2021-05-19:
Which version should I use? I was using the master branch. Right now I'm trying to setup release 0.3. I'll let you know it that works for me.

jdavidberger, 2021-05-19:
Master I think; lots of recent changes and a new tag is probably coming soon. Those failures might be related to performance of the system; make sure you are compiled as Release or RelWithDebInfo?

CarlosDanielFP, 2021-05-19:
You are right. I tried using a Jetson board and it works great. I'll deal with the Raspberry some other day.
Thank you so much for the prompt answers and confirmation that the new trackers actually work with the current state of the library.
If there is anything on the entry level that you need help with, I can try to contribute. This library is being really helpful.

jdavidberger, 2021-05-19:
No problem. There are performance tuning flags you can use to maybe make it better but compiling in release mode and making sure it's using native libraries is important for the RPI. Join us at https://discordapp.com/invite/7QbCAGS if you have any other questions. Always appreciate the testing / feedback!

closed 2021-05-19 by jdavidberger