collabora/libsurvive #238: CMake fails to detect roscpp and geometry_msgs CMake files without further modification

happysmash27, 2021-07-04:
**Describe the bug**
The default Makefile script will not detect roscpp or geometry_msgs. I need to manually do the same thing it would do to compile. 

**Compile log**
Without any modification: 
```
libsurvive-testneeded % make
mkdir -p bin
cd bin && cmake .. && cmake --build .
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_GZVPRINTF
-- Performing Test HAVE_GZVPRINTF - Success
-- Looking for pthread.h
CMake Warning (dev) at /usr/share/cmake/Modules/CheckIncludeFile.cmake:80 (message):
  Policy CMP0075 is not set: Include file check macros honor
  CMAKE_REQUIRED_LIBRARIES.  Run "cmake --help-policy CMP0075" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  CMAKE_REQUIRED_LIBRARIES is set to:

    z

  For compatibility with CMake 3.11 and below this check is ignoring it.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindThreads.cmake:142 (CHECK_INCLUDE_FILE)
  CMakeLists.txt:101 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Looking for gattlib.h
-- Looking for gattlib.h - not found
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Looking for X11/Xlib.h
-- Looking for X11/Xlib.h - found
Found cblas at /usr/lib64/libcblas.so
Found openblas at /usr/lib64/libopenblas.so
Found lapacke at /usr/lib64/liblapacke.so
Searching for 'cblas.h'; found at /usr/include/cblas.h
-- Looking for lapacke/lapacke.h
-- Looking for lapacke/lapacke.h - not found
Using blas backend /usr/lib64/libcblas.so
-- Looking for fopencookie
-- Looking for fopencookie - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for libusb/libusb.h
-- Looking for libusb/libusb.h - not found
Using versioned libusb dir
-- Using CATKIN_DEVEL_PREFIX: /home/happysmash27/Gits/Test/libsurvive-testneeded/bin/devel
-- Using CMAKE_PREFIX_PATH:
-- Found PythonInterp: /usr/bin/python (found version "3.8.4")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.8/site-packages/em.py
-- Using empy: /usr/lib/python3.8/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/happysmash27/Gits/Test/libsurvive-testneeded/bin/test_results
-- Found gmock: gmock and gtests will be built
-- nosetests not found, Python tests can not be run (try installing package 'python3-nose')
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
CMake Warning (dev) at tools/ros_publisher/CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find roscpp (missing: roscpp_DIR)
-- Could not find the required component 'roscpp'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:88 (find_package):
  Could not find a package configuration file provided by "roscpp" with any
  of the following names:

    roscppConfig.cmake
    roscpp-config.cmake

  Add the installation prefix of "roscpp" to CMAKE_PREFIX_PATH or set
  "roscpp_DIR" to a directory containing one of the above files.  If "roscpp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  tools/ros_publisher/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/happysmash27/Gits/Test/libsurvive-testneeded/bin/CMakeFiles/CMakeOutput.log".
See also "/home/happysmash27/Gits/Test/libsurvive-testneeded/bin/CMakeFiles/CMakeError.log".
make: *** [Makefile:5: build] Error 1
```

Running CMake manually to change roscpp_DIR: 

```
bin % cmake .. -Droscpp_DIR=/usr/share/roscpp/cmake
Found cblas at /usr/lib64/libcblas.so
Found openblas at /usr/lib64/libopenblas.so
Found lapacke at /usr/lib64/liblapacke.so
Searching for 'cblas.h'; found at /usr/include/cblas.h
Using blas backend /usr/lib64/libcblas.so
Using versioned libusb dir
-- Using CATKIN_DEVEL_PREFIX: /home/happysmash27/Gits/Test/libsurvive-testneeded/bin/devel
-- Using CMAKE_PREFIX_PATH:
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Using empy: /usr/lib/python3.8/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/happysmash27/Gits/Test/libsurvive-testneeded/bin/test_results
-- Found gmock: gmock and gtests will be built
-- nosetests not found, Python tests can not be run (try installing package 'python3-nose')
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
CMake Warning (dev) at tools/ros_publisher/CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find geometry_msgs (missing: geometry_msgs_DIR)
-- Could not find the required component 'geometry_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:88 (find_package):
  Could not find a package configuration file provided by "geometry_msgs"
  with any of the following names:

    geometry_msgsConfig.cmake
    geometry_msgs-config.cmake

  Add the installation prefix of "geometry_msgs" to CMAKE_PREFIX_PATH or set
  "geometry_msgs_DIR" to a directory containing one of the above files.  If
  "geometry_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  tools/ros_publisher/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/happysmash27/Gits/Test/libsurvive-testneeded/bin/CMakeFiles/CMakeOutput.log".
See also "/home/happysmash27/Gits/Test/libsurvive-testneeded/bin/CMakeFiles/CMakeError.log".
```

including geometry_msgs with `cmake .. -Droscpp_DIR=/usr/share/roscpp/cmake -Dgeometry_msgs_DIR=/usr/share/geometry_msgs/cmake` finally works to configure it. This is not an ideal situation as it bypasses the normal make script, and I'm not sure if this is the proper way to do it. 

**Desktop (please complete the following information):**
 - OS: Gentoo Linux

happysmash27, 2022-08-02:
New problem: 
```
-- Could NOT find tf (missing: tf_DIR)
-- Could not find the required component 'tf'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:88 (find_package):
  Could not find a package configuration file provided by "tf" with any of
  the following names:

    tfConfig.cmake
    tf-config.cmake

  Add the installation prefix of "tf" to CMAKE_PREFIX_PATH or set "tf_DIR" to
  a directory containing one of the above files.  If "tf" provides a separate
  development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  tools/ros_publisher/CMakeLists.txt:11 (find_package)
```

New solution: 
```
cmake .. -DCMAKE_INSTALL_PREFIX="/usr/local" -DCMAKE_BUILD_TYPE=RelWithDebInfo -Droscpp_DIR=/usr/share/roscpp/cmake -Dgeometry_msgs_DIR=/usr/share/geometry_msgs/cmake -Dtf_DIR=/usr/share/tf/cmake
```

nmerrill67, 2023-11-21:
@ happysmash27 I experienced the same problem with my custom package. For some reason my package can not find roscpp but others in my catkin workspace can. Now it cannot build without specifying roscpp_DIR and similar for all other ros package depends (my catkin build command is very long now). If you have any additional insight as to why this happened please let me know.