collabora/libsurvive #240: Building Win10 VS2019

rygo6, 2021-08-11:
I am trying to build on Win10 with the power shell script, but it fails with a handfull of errors. I believe it is using VS2019 to compile.

It runs into about 20 errors like this:
```
C:\Developer\libsurvive\src\survive_kalman_tracker.c(24,1): error C2059: syntax error: 'type' [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
```

Here is the full log:
```
PS C:\Developer\libsurvive> .\make.ps1
DEBUG:    1+  >>>> .\make.ps1
DEBUG:    1+  >>>> Set-PSDebug -Trace 1
DEBUG:    2+  >>>> $flag=$args[0]
DEBUG:    4+ if( >>>> $flag -eq "--nuget") {
DEBUG:   28+   >>>> md build-win -ea 0
DEBUG:   36+ begin  >>>> {
DEBUG:   39+          >>>> $wrappedCmd = $ExecutionContext.InvokeCommand.GetCommand('New-Item', [System.Management.Automation.CommandTypes]::Cmdlet)
DEBUG:   40+          >>>> $scriptCmd = {& $wrappedCmd -Type Directory @PSBoundParameters }
DEBUG:   41+          >>>> $steppablePipeline = $scriptCmd.GetSteppablePipeline()
DEBUG:   42+          >>>> $steppablePipeline.Begin($PSCmdlet)
DEBUG:   47+  >>>> }
DEBUG:   49+ process  >>>> {
DEBUG:   52+          >>>> $steppablePipeline.Process($_)
DEBUG:   57+  >>>> }
DEBUG:   59+ end  >>>> {
DEBUG:   62+          >>>> $steppablePipeline.End()
DEBUG:   67+  >>>> }
DEBUG:   29+   >>>> cd build-win
DEBUG:   30+   >>>> cmake ..
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19043.
Could not find cblas
Could not find openblas
Could not find lapacke
Using blas backend openblas
Nuget is found at C:/Program Files (x86)/NuGet/Visual Studio 2017/nuget.exe
C:/Program Files/dotnet/dotnet.exe build -c Release
Microsoft (R) Build Engine version 16.7.0+b89cb5fde for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  All projects are up-to-date for restore.
  libsurvive.net -> C:\Developer\libsurvive\bindings\cs\libsurvive.net\bin\Release\netstandard2.0\libsurvive.net.dll
  Demo -> C:\Developer\libsurvive\bindings\cs\Demo\bin\Release\netcoreapp2.1\Demo.dll

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:01.03
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Developer/libsurvive/build-win
DEBUG:   31+   >>>> cmake --build . --config Release
Microsoft (R) Build Engine version 16.7.0+b89cb5fde for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/Developer/libsurvive/redist/CMakeLists.txt
  CNGFX.vcxproj -> C:\Developer\libsurvive\build-win\redist\Release\CNGFX.lib
  Building Custom Rule C:/Developer/libsurvive/redist/CMakeLists.txt
  mpfit.vcxproj -> C:\Developer\libsurvive\build-win\redist\Release\mpfit.lib
  Restoring nuget dependencies
  MSBuild auto-detection: using msbuild version '16.7.0.37604' from 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Current\Bin'.
  All packages listed in packages.config are already installed.
  Building Custom Rule C:/Developer/libsurvive/redist/CMakeLists.txt
  survive_matrix.vcxproj -> C:\Developer\libsurvive\build-win\redist\Release\survive_matrix.lib
  Building Custom Rule C:/Developer/libsurvive/src/CMakeLists.txt
  ootx_decoder.c
  poser.c
C:\Developer\libsurvive\redist\sv_matrix.h(217,26): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
[... 129 lines cut ...]
C:\Developer\libsurvive\src\lfsr.c(82,30): warning C4334: '<<': result of 32-bit shift implicitly converted to 64 bits (was 64-bit shift intended?) [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
  lfsr_lh2.c
  survive_str.c
  str.c
  survive_async_optimizer.c
  linmath.c
C:\Developer\libsurvive\redist\sv_matrix.h(217,26): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\sv_matrix.h(217,29): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1065,1): warning C4267: 'initializing': conversion from 'size_t' to 'int16_t', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1066,1): warning C4267: 'initializing': conversion from 'size_t' to 'int16_t', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1110,1): warning C4267: 'initializing': conversion from 'size_t' to 'int16_t', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1157,22): warning C4267: '=': conversion from 'size_t' to 'int16_t', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1167,21): warning C4267: '=': conversion from 'size_t' to 'int16_t', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1182,2): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1195,2): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
C:\Developer\libsurvive\redist\linmath.c(1196,2): warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data [C:\Developer\libsurvive\build-win\src\survive.vcxproj]
  puff.c
  symbol_enumerator.c
  jsmn.c
  json_helpers.c
  crc32.c
  hid-windows.c
  getdelim.c
  Generating Code...
```

nicholasAWwright, 2021-09-22:
I am also having this issue with the same console output on two separate computers running Windows10 and VS2019.

BenWoodford, 2021-09-23:
There’s an issue with the macros in play, I was having a look through it yesterday but VS is refusing to output the preprocessed file for me so I can’t actually see the end result of the preprocessor directives…

BenWoodford, 2021-09-29:
Having a trawl through the preprocessed file I managed to get to spit out, and the troublesome output is as follows:

`_Generic((&t->light_variance), double*: survive_attach_configf, float*: survive_attach_configf, int*: survive_attach_configi )(ctx, "lh-light-variance", &t->light_variance);`

My C is rusty and `_Generic` call isn't something I've used before if I'm honest.

There's also another error in `survive_optimizer.c` at line 858 due to `measurements_buffer` not having a size, which spits ou:

`survive_optimizer.c(858,66): error C2036: 'void *': unknown size`

jdavidberger, 2021-12-19:
I think this should be resolved now; although I think it will only work for vs2019 and later.

closed 2021-12-19 by jdavidberger