collabora/libsurvive #247: Time inconsistancy in events

bl4ckb0ne, 2021-12-10:
**Describe the bug**
There seems to be time inconsistancy in the events, pose and button are not based on the same time. One is using `OGStartTimeS` and the other one `timestamp_in_s`, which are slightly different.

**Data**

Here's a print from my application, the first line commes from the time in a `struct SurviveSimpleButtonEvent`, the second one from a `struct SurviveSimplePoseUpdatedEvent`

```
process button event time 8.015319
process pose event time 1639169164.974834
```

**Desktop (please complete the following information):**
 - OS: Linux
 - Version `master`

**Additional context**
I don't understand if this is wanted or if this is an error, I can provide a patch for the latter.

jdavidberger, 2021-12-16:
This isn't desired behavior; I'd say the buttons should be in global time. I'd gladly merge in a PR to this effect!

bl4ckb0ne, 2021-12-16:
Good to know, I'll send a patch tomorrow.

PR #250 (2021-12-17): survive_api: button event now uses global time
Fixes #247
closed 2021-12-17 by jdavidberger