collabora/libsurvive #252: Null pointer crash in driver_vive.c

blitz-research, 2021-12-23:
Hi,

I'm new to libsurvive and am trying to get it going with 'monado', the openxr runtime.

I was initially using monado's built-in 'vive' driver, but this was unable to detect my lighthouses for some reason. Also, the left/right eyes were swapped or something but that's probably a different issue...

I then found libservive and got it building OK, but I now get a crash when trying to use it, eg: with a plain 'monado-cli probe' or trying to start monado-service etc.

So I made a debug build of libsurvive and can now see where it's crashing due to a null pointer, but not why - it's line 794 of driver_vive.c:

791│         // There should only be one HMD, tie the mainboard interface to the surviveobject
792│         if (info->type == USB_DEV_HMD_IMU_LH || info->type == USB_DEV_HMD) {
793│                 if (sv->hmd_imu_index != -1 && sv->hmd_mainboard_index != -1) {
794├───────────────────────> sv->udev[sv->hmd_mainboard_index]->so = sv->udev[sv->hmd_imu_index]->so;
795│                         usbInfo->ownsObject = false;
796│                 }
797│         }

Looking at the locals, sv->hmd_imu_index is '5', but the fifth entry of the sv->udev[] array is null.

Please fix this as it'd be nice to be able to try libsurvive!

blitz-research, 2021-12-23:
OK, it might be in line 748:

    struct SurviveUSBInfo *usbInfo = sv->udev[sv->udev_cnt++] = SV_CALLOC(sizeof(struct SurviveUSBInfo));

This bumps the udev_cnt var, but this same var is later assigned to some index vars (lines 764 and 770) which might mean this indices are 'one off'? Also, the SV_CALLOC looks like it leaks if opening usb device fails.

I tried assigning udev_cnt-1 to index vars instead and it stopped libsurvive crashing, but 'monado-cli probe' now never returns, although it looks like it succeeds. I also get no HMD output, where-as I used to with the monado 'vive' driver.

blitz-research, 2021-12-23:
Yikes, tried again with my udev_cnt-1 hack above again and it's working!

Well, sort of - the steroscopic effect is working fine, but I get some nasty 'juddering' when I rotate my head. Also, there a weird 'vsync' style black horizontal line slowly scrolling up the display!

jdavidberger, 2021-12-29:
Should be fixed now, thanks for reporting this!

commit 18993e72 (2021-12-29): Fixed #252
closed 2021-12-29 by jdavidberger