collabora/libsurvive #255: New release

PureTryOut, 2022-01-03:
I was wondering if a new release could be made. The last one was on the 22nd of October 2020 which is _more_ than a year ago and it seems _a lot_ of improvements have been made since which users have been missing out on.

blitz-research, 2022-01-03:
I agree, I've been looking at some of the issues in the hope of helping out a bit but there's been a lot of changes to some of the key source files, esp. survive_optimize.c which seems important, hopefully for the better!

jdavidberger, 2022-01-03:
Definitely warranted; the things blocking this is there is an oustanding USB issue I want to fix and get the groundtruth testing files in order. Hopefully will get it in today

jdavidberger, 2022-02-06:
New version released!

closed 2022-02-06 by jdavidberger