collabora/libsurvive #258: Vive_Pro 2 MV

CygnusX1111, 2022-01-20:
**Describe the bug**
is missing in device list.

**Data**
Error loading /develop/VR/libsurvive/libsurvive-master/build/plugins/driver_usbmon.so: /usr/lib64/liblapacke.so.3: undefined symbol: cunmlq_
Info: Adding tracked object WM0 from HTC
Info: Adding tracked object T20 from HTC
Info: All enumerated devices attached.
Info: Loaded drivers: HTCVive
Info: Device WM0 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x81020006 Board rev: 3 (len 64)
Warning: Unknown model_number 'VIVE Controller Pro MV'. Please submit an issue with this value describing your device so it can be added to the known list.
Info: Detected LH gen 2 system.
Info: LightcapMode (WM0) 1 -> 2 (ff)
Info: Device T20 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x85020006 Board rev: 3 (len 64)
Warning: Unknown model_number 'VIVE_Pro 2 MV'. Please submit an issue with this value describing your device so it can be added to the known list.
Info: T20 is treated as HMD device
Info: LightcapMode (T20) 1 -> 2 (4)
Info: OOTX not set for LH in channel 1; attaching ootx decoder using device WM0
Warning: Fail to connect to the bluetooth device D8:0B:1E:A9:85:0E.
Info: OOTX not set for LH in channel 0; attaching ootx decoder using device WM0
Warning: Fail to connect to the bluetooth device FD:89:9C:6B:FC:55.

**Desktop (please complete the following information):**
 - OS: Linux OpenSUSE Leap
 - Version 15.2

**Additional context**
After a lot of patching cmake files, i managed to launch survive-cli.
Ther also seems a mismatch in cnkalman_extrapolate_state() (4 arguments vs. 5 args in survive_kalman_tracker.c)
I will investigate further...

jdavidberger, 2022-01-21:
Added in latest push. It sounds like you need to run `git submodule update --init`; that API did change but AFAICT it's tracking the right submodule commit for it. 

If not definitely let me know though!

closed 2022-01-21 by jdavidberger

CygnusX1111, 2022-01-22:
Great thanks a lot. At least i can compile and run.
But api_examle --v 2 shows some unknown data:
`
(  0.000) SimpleApi: Available Posers:
(  0.000) SimpleApi:     BaryCentricSVD
(  0.000) SimpleApi:     Dummy
(  0.000) SimpleApi:    *MPFIT
(  0.000) SimpleApi:     KalmanOnly
(  0.000) SimpleApi: Totals 4 Posers.
(  0.000) SimpleApi: Using 'MPFIT' for poser
(  0.000) SimpleApi: Available Disambiguators:
(  0.000) SimpleApi:    *StateBased
(  0.000) SimpleApi: Totals 1 Disambiguators.
(  0.000) SimpleApi: Using 'StateBased' for disambiguator
(  0.040) SimpleApi: Adding tracked object WM0 from HTC
(  0.052) SimpleApi: Adding tracked object T20 from HTC
(  0.052) SimpleApi: All enumerated devices attached.
(  0.052) SimpleApi: Loaded drivers: HTCVive
Found 'WM0'
Found 'T20'
Found 'LH0'
Found 'LH1'
(1642836975.242761) Found 'WM0'
(1642836975.254718) Found 'T20'
(1642836975.254965) Found 'LH0'
(1642836975.254965) Found 'LH1'
(  0.914) SimpleApi: Device WM0 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x81020006 Board rev: 3 (len 64)
(1642836976.116664) WM0 received configuration of length 9318 type 3-2
(  1.326) SimpleApi: Fail to connect to the bluetooth device D8:0B:1E:A9:85:0E.
(  1.351) SimpleApi: Detected LH gen 2 system.
(  1.354) SimpleApi: LightcapMode (WM0) 1 -> 2 (ff)
(  1.426) SimpleApi: Device T20 has watchman FW version 1526689969 and FPGA version 531/7/2; named '                       watchman'. Hardware id 0x85020006 Board rev: 3 (len 64)
(  1.426) SimpleApi: T20 is treated as HMD device
(1642836976.628992) T20 received configuration of length 11924 type 2-1
(  1.447) SimpleApi: LightcapMode (T20) 1 -> 2 (4)
(  1.448) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xa3)

a3 b5   90 0d 
(  1.458) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x63)

63 e1   12 0e 
(  1.468) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x33)

33 0d   95 0e 
(  1.477) SimpleApi: Fail to connect to the bluetooth device FD:89:9C:6B:FC:55.
(  1.480) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xd3)

d3 42   17 0f 
(  1.481) SimpleApi: OOTX not set for LH in channel 1; attaching ootx decoder using device WM0
(  1.490) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xd3)

d3 70   99 0f 
(  1.501) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x33)

33 9e   1b 00 
(  1.512) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xf3)

f3 ca   9d 00 
(  1.523) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xf3)

f3 f7   1f 01 
(  1.535) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x73)

73 24   a2 01 
(  1.546) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xa3)

a3 51   24 02 
(  1.547) SimpleApi: OOTX not set for LH in channel 0; attaching ootx decoder using device WM0
(  1.558) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xd3)

d3 7d   a6 02 
(  1.568) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xb3)

b3 a9   28 03 
(  1.581) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xd3)

d3 d4   aa 03 
(  1.590) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xc3)

c3 02   2d 04 
(  1.601) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xa3)

a3 2d   af 04 
(  1.612) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x53)

53 61   31 05 
(  1.623) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x73)

73 8b   b3 05 
(  1.635) SimpleApi: Not entirely sure what this data is; errors may occur (5, 0x73)

11 ac   b9 6c 01 73   b5 35 06 
(  1.646) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xc3)

c3 e8   b7 06 
(  1.657) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xe3)

e3 12   3a 07 
(  1.668) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x93)

93 3f   bc 07 
(  1.680) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xf3)

f3 6b   3e 08 
(  1.690) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x63)

63 9f   c0 08 
(  1.701) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x23)

23 ca   42 09 
(  1.712) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xa3)

a3 f6   c4 09 
(  1.723) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x43)

43 25   47 0a 
(  1.735) SimpleApi: Not entirely sure what this data is; errors may occur (5, 0xd3)

11 fc   c5 90 06 d3   52 c9 0a 
(  1.746) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xf3)

f3 7f   4b 0b 
(  1.757) SimpleApi: Not entirely sure what this data is; errors may occur (5, 0x43)

01 a8   25 db 06 43   ac cd 0b 
(  1.768) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xe3)

e3 d7   4f 0c 
(  1.780) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x63)

63 02   d2 0c 
(  1.791) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xd3)

d3 34   54 0d 
(  1.802) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xc3)

c3 61   d6 0d 
(  1.813) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0x33)

33 8e   58 0e 11 24   6a 7a 03 
(  1.823) SimpleApi: Not entirely sure what this data is; errors may occur (0, 0xc3)

`

...

jdavidberger, 2022-01-22:
So the device is sending unknown opcodes back. Since I don't have a vive pro; I can't fix this without the data. If you can capture a data packet, there is a pretty good chance I can fix it though. Here are instructions on how to do that: https://github.com/cntools/libsurvive#raw-usb-recording

If you can record 30ish seconds of the device just sitting in one spot, and send me that file, I'll take a look at it.

CygnusX1111, 2022-01-22:
First of all good news, survive-cli -v 200 shows no more warnings of unknown data.
I made a room-calibration then i dropped SteamVR.
Then i started again with Monado(my local patch without survive) and direct mode rendering works(with wrong display parameters at the moment).
I recognized that writing to HID Config failed in Broken pipe(like SteamVR).
I tried to adjust the packet sizes, but nada...
While ignoring unsuccessful writing Monado continues working as OpenXR.

It seems survive is not ignoring those write errors:
(one Vive Controller1.5 marked Tracker)
I will continue my investigations tomorrow..

Info: Tracker config for T20 (16 state count)
Info: Adding tracked object T20 from HTC
Info: Requesting config for T20 (nil) 0
Info: USB device 046d:c31c4x in an unknown type; ignoring
Info: USB device 046d:c0774x in an unknown type; ignoring
Info: All enumerated devices attached.
Info: Loaded drivers: HTCVive
Info: Position found for LH 0(ID: dae2d519, mode:  1, err: 0.008023) +1.423225e+00   -7.312127e-01   +2.057659e+00      +8.267592e-01   +4.072700e-01   +1.194750e-01   +3.692237e-01
Info: Position found for LH 1(ID: 5947b229, mode:  0, err: 0.034110) -3.687149e-01   +2.186163e+00   +2.021577e+00      +2.671867e-01   +2.864332e-01   -4.985626e-01   -7.733063e-01
Info: Submitting config for WM0 Watchman at 0.028033
Info: Waiting, Transfer status -2 at 0.014805 sec for WM0 (0)
Info: Submitting config for T20 Tracker (2018) at 0.042150
Info: Waiting, Transfer status -2 at 0.000838 sec for T20 (0)
Info: Submitting config for WM0 Watchman at 0.062845
Info: Waiting, Transfer status -2 at 0.035031 sec for WM0 (0)
Info: Submitting config for T20 Tracker (2018) at 0.062989
Info: Config readmode in 0.021193 sec for T20 expected length: 3752
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.021361 sec for T20 (0)
Info: Submitting config for WM0 Watchman at 0.083064
Info: Waiting, Transfer status -2 at 0.055312 sec for WM0 (0)
Info: Submitting config for T20 Tracker (2018) at 0.083514
Info: Config response in 0.041666 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.041834 sec for T20 (0)
Info: Submitting config for WM0 Watchman at 0.103348
Info: Waiting, Transfer status -2 at 0.075744 sec for WM0 (0)
Info: Submitting config for T20 Tracker (2018) at 0.103985
Info: Config response in 0.062184 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.062412 sec for T20 (0)
Info: Submitting config for WM0 Watchman at 0.123781
Info: Waiting, Transfer status -2 at 0.096068 sec for WM0 (0)
Info: Submitting config for T20 Tracker (2018) at 0.124564
Info: Config response in 0.082747 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.082994 sec for T20 (0)
Info: Submitting config for WM0 Watchman at 0.144103
Info: Waiting, Transfer status -2 at 0.116471 sec for WM0 (0)
Info: Submitting config for T20 Tracker (2018) at 0.145147
Info: Config response in 0.103351 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.103662 sec for T20 (0)
Info: Submitting config for T20 Tracker (2018) at 0.165815
Info: Config response in 0.124021 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.124265 sec for T20 (0)
Info: Submitting config for T20 Tracker (2018) at 0.186422
Info: Config response in 0.144709 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.144832 sec for T20 (0)
Info: Submitting config for T20 Tracker (2018) at 0.206985
Info: Config response in 0.165107 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.165364 sec for T20 (0)
Info: Submitting config for T20 Tracker (2018) at 0.227521
Info: Config response in 0.185733 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.185983 sec for T20 (0)
Info: Submitting config for T20 Tracker (2018) at 0.248140
Info: Config response in 0.206357 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000
Info: Submitting config for T20 Tracker (2018) at 0.000000
Info: Waiting, Transfer status -2 at 0.206636 sec for T20 (0)
Info: Submitting config for T20 Tracker (2018) at 0.268789
Info: Config response in 0.226988 sec for T20 17 62
Info: Resubmit startup packet for T20 Tracker (2018) at 0.000000

[ConfigReadJson.zip](https://github.com/cntools/libsurvive/files/7919254/ConfigReadJson.zip)

CygnusX1111, 2022-01-24:
Hi jdavid,
 bluetooth is now working, had to install Broadcom patch.
Looks better now, but in monado-service i get those unknown packet messages again.

[Logs.zip](https://github.com/cntools/libsurvive/files/7925318/Logs.zip)

Green0Photon, 2022-02-21:
I recently tried this on my VP2 on Windows. This is on 1.01, 1.0 didn't work at all, so thankfully those bug fixes worked. 

Procedure for launch: first I turned the link box on, which auto opens Vive Console and SteamVR. I hurriedly close out of these, sometimes they open again. I turned on the lighthouses with an Android app. Then, I can run libsurvive.

Interestingly enough, mine actually showed something slightly different at the beginning.

```
Info: Device T20 has watchman FW version 1526689969 and FPGA version 531/7/2; named ' watchman'. Hardware id 0x85020006 Board rev: 3 (len 65)
Warning: Unknown model_number 'VIVE_Pro 2 PV'. Please submit an issue with this value describing your device so it can be added to the known list.
Info: Read config for T20
Info: Detected LH gen 2 system.
Info: LightcapMode (T20) 1 -> 2 (4)
Info: OOTX not set for LH in channel 0; attaching ootx decoder using device T20
Info: OOTX not set for LH in channel 1; attaching ootx decoder using device T20
(And so on, lighthouse calculations.)
```

Besides that bit of weirdness, the headset seems to be tracked perfectly fine! I can see the headset in the gui, the trails, each sensor being picking up one or the other or both lighthouses. Very cool.

However, when I do this, it doesn't detect controllers at all (I'm using Index knuckles). For this, there are two different procedures. One is letting SteamVR launch and turning off the controller and base station power management stuff, so that the controllers remain connected and green, and the base stations remain on and green. When I do this, same logs as above appear.

The other option is turning on SteamVR. Which leads to a very similar set of unknown packets as @CygnusX1111. The headset itself still tracks perfectly fine, but nothing with the controllers connect.

Logs here: [logs.txt](https://github.com/cntools/libsurvive/files/8106352/logs.txt)

Maybe later I'll try doing some packet capture with usbpcap to see if that will help out.