collabora/libsurvive #263: Tracking very jittery

puckipedia, 2022-02-14:
**Describe the bug**
I've been trying to use monado + libsurvive, and it seems that the tracking is /very/ jittery. I can properly replicate this with just libsurvive on its own, using tagged v1.0 (though playing back the data in 1.01 gives the same result).

**Data**
I can't upload the gz (or zipped) to Github, so it's currently located at https://puck.moe/tmp/walk.rec.gz. and https://puck.moe/tmp/walk.rec.gz.json
This is me using the controllers (connected via the HMD, out of view of one base station, previously calibrated), and walking clockwise a bunch. At some point I put down the controllers, closed the curtains (to make sure the reflections from the window aren't interfering), walked some more circles (not all an identical path). The lighthouses are probably slightly weirdly placed, but this happened identically on previous less experimental lighthouse placements, which worked fine on SteamVR.

**Hardware setup**
I'm using a very recent (<1mo old) Valve Index, using the Gen2 lighthouses.

**Desktop (please complete the following information):**
 - OS: Linux (NixOS)
 - Version: Kernel 5.12.9

**Additional context**
(Thank you for the great tooling/library, btw! Very excited to make use of it :p)

jdavidberger, 2022-02-14:
The data coming out of this looks reasonable; but I can see that the initial LH calibration is very broken. If it's possible to run the same test with openvr:

https://github.com/cntools/libsurvive/wiki/Comparing-to-OpenVR

That will help me narrow down what is going on since right now I have no confidence in the LH positions coming out of this recording so I'm not sure in which direction to start looking into it. If that isn't doable though I can still probably make it work.

puckipedia, 2022-02-14:
I'll give that a try, but it'll be a bit; i'm currently running Steam in a flatpak and getting data out of it is non-trivial.. I'm planning to un-flatpak Steam soon, but it'll probably be a few days.