collabora/libsurvive #264: Bulding libsurvive on windows

mgavornik, 2022-02-25:
Hi, when I try to run the .\make.ps1 script on windows I get the errors 
```

CMake Error at CMakeLists.txt:196 (add_subdirectory):
The source directory
    D:/b/master/library/libsurvive_latest/libs/cnkalman
does not contain a CMakeLists.txt file.
  ...
CMake Error at src/CMakeLists.txt:14 (cnkalman_generate_code):
Unknown CMake command "cnkalman_generate_code".
```

Am I doing something wrong ?

jdavidberger, 2022-02-25:
That directory is a submodule, so you have to do a submodule init and update -- `git submodule update --init --recursive`. 

I get this question enough that I might just have a script do it if the user doesn't

mgavornik, 2022-02-25:
This worked, thank you very much David.

closed 2022-02-25 by mgavornik