collabora/libsurvive #266: T20 Device turned off

reyrael, 2022-03-04:
I originally started with v0.3 on a raspberry pi running the python example code to get the pose from a VIVE Tracker and I upgraded to v1.1.3. I am now getting the `T20 Device turned off` output below with ```python full-example.py --lighthouse-gen 2 --force-calibrate 1 --haptic-on-calibrate 1  --lighthousecount 2 --v 1000```

```[0.03927446092854493, 0.003942799043242111, 0.9839841733142239, 0.0005320908832763885, 0.013885842749911503, -0.007711033635954323, 0.0, 0.0, 0.0]
Info: T20 IMU bdcc05e4 ( 66.339): +3.927446e-02   +3.942799e-03   +9.839842e-01   +5.320909e-04   +1.388584e-02   -7.711034e-03
Info: T20 IMU 3.757117 66.342851  4
[0.04366503722992034, -0.0014336660514993227, 0.9907743101744418, 0.0037278841913714732, 0.013885842749911503, -0.00558050476389093, 0.0, 0.0, 0.0]
Info: T20 IMU bdcef49b ( 66.343): +4.366504e-02   -1.433666e-03   +9.907743e-01   +3.727884e-03   +1.388584e-02   -5.580505e-03
Warning: 3.760867 T20 Device turned off: 2
Info: Closing device on T20 (nil) (0x1e9c93b0)
Info: Cleaning up interface on 7 T20 IMU (0x1e9cade0)
Info: Cleaning up interface on 11 T20 Lightcap (0x1e9caed0)
Info: Cleaning up interface on 15 T20 Buttons (0x1e9cafc0)
Info: Cleaning up transfer on 15 Buttons
Info: Cleaning up transfer on 7 IMU
Info: Cleaning up transfer on 11 Lightcap
Info: Requesting close for unknown
Info: Closing device T20 Tracker (2018) (0/1)
Info: Disconnecting device T20 at 3.7616900
```

more output
```[0.04195759089049657, -0.00021174216633081499, 0.9844691830899538, 0.0026626197553397783, 0.012820578313879803, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.042201511796128535, 0.002965259935107305, 0.9893192808472523, 0.0026626197553397783, 0.013885842749911503, -0.008776298071986016, 0.0, 0.0, 0.0]
[0.03927446092854493, 0.004920338151376917, 0.9832566586506292, 0.0005320908832763885, 0.01175531387784811, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.039030540022912966, 0.0032096447121410065, 0.9815591244355747, 0.0015973553193080834, 0.01175531387784811, -0.00558050476389093, 0.0, 0.0, 0.0]
[0.0424454327017605, 0.001987720826972499, 0.9820441342113045, 0.0005320908832763885, 0.014951107185943197, -0.0045152403278592365, 0.0, 0.0, 0.0]
[0.04268935360739247, 0.0022321056040062004, 0.9864092221928732, 0.0005320908832763885, 0.012820578313879803, -0.0045152403278592365, 0.0, 0.0, 0.0]
[0.04049406545670477, -0.00045612694336451655, 0.9878642515200627, -0.0005331735527553082, 0.012820578313879803, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.04268935360739247, 0.004920338151376917, 0.9893192808472523, 0.0026626197553397783, 0.013885842749911503, -0.0045152403278592365, 0.0, 0.0, 0.0]
[0.0417136699848646, 0.002476490381039902, 0.9888342710715224, 0.0015973553193080834, 0.014951107185943197, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.0395183818341769, 0.005164722928410618, 0.980589104884115, 0.0005320908832763885, 0.01601637162197489, -0.00558050476389093, 0.0, 0.0, 0.0]
Warning: 2.698042 T20 Device turned off: 2
Info: MPFIT stats for T20:
Info: 	seed runs         0 / 0
Info: 	error failures    0
Info: Adding tracked object T20 from HTC
Info: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
Info: LightcapMode (T20) 1 -> 2 (4)
[0.04342111632428837, 0.004675953374343215, 0.986894231968603, -0.0005331735527553082, 0.013885842749911503, -0.0045152403278592365, 0.0, 0.0, 0.0]
[0.042201511796128535, 0.0032096447121410065, 0.9851966977535486, -0.001598437988787003, 0.013885842749911503, -0.00558050476389093, 0.0, 0.0, 0.0]
[0.04366503722992034, 0.002965259935107305, 0.9849541928656836, 0.0015973553193080834, 0.010690049441816417, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.039030540022912966, 0.00540910770544432, 0.9847116879778187, -0.0005331735527553082, 0.014951107185943197, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.042201511796128535, -0.00021174216633081499, 0.9854392026414135, 0.0005320908832763885, 0.014951107185943197, -0.00558050476389093, 0.0, 0.0, 0.0]
[0.035859568249697396, 0.005164722928410618, 0.9873792417443329, -0.001598437988787003, 0.013885842749911503, -0.006645769199922626, 0.0, 0.0, 0.0]
[0.03781093549475313, 0.0032096447121410065, 0.9847116879778187, 0.0015973553193080834, 0.013885842749911503, -0.0045152403278592365, 0.0, 0.0, 0.0]
```
 api_example
```
~/libsurvive $ ./bin/api_example 
(  0.001) SimpleApi: Loaded drivers: GlobalSceneSolver, HTCVive
Found 'LH0'
(  0.031) SimpleApi: Adding tracked object T20 from HTC
(1646422733.427387) Found 'LH0'
(1646422733.455947) Found 'T20'
(  0.794) SimpleApi: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
(1646422734.219460) T20 received configuration of length 8070 type 3-7
(  0.820) SimpleApi: Detected LH gen 2 system.
(  0.821) SimpleApi: Adding lighthouse ch 2 (idx: 0, cnt: 1)
(  0.821) SimpleApi: OOTX not set for LH in channel 2; attaching ootx decoder using device T20
(  0.837) SimpleApi: LightcapMode (T20) 1 -> 2 (4)
(  1.824) SimpleApi: 1.823924 T20 Device turned off: 2
(  1.825) SimpleApi: MPFIT stats for T20:
(  1.825) SimpleApi: 	seed runs         0 / 0
(  1.825) SimpleApi: 	error failures    0
(  1.859) SimpleApi: Adding tracked object T20 from HTC
(1646422735.283620) Found 'T20'
(  2.622) SimpleApi: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
(1646422736.046982) T20 received configuration of length 8070 type 3-7
(  2.669) SimpleApi: LightcapMode (T20) 1 -> 2 (4)
(  2.819) SimpleApi: OOTX not set for LH in channel 0; attaching ootx decoder using device T20
( 10.209) SimpleApi: 10.209045 T20 Device turned off: 2
( 10.210) SimpleApi: MPFIT stats for T20:
( 10.210) SimpleApi: 	seed runs         0 / 0
( 10.210) SimpleApi: 	error failures    0
( 10.235) SimpleApi: Adding tracked object T20 from HTC
(1646422743.660215) Found 'T20'
( 11.016) SimpleApi: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
(1646422744.441473) T20 received configuration of length 8070 type 3-7
( 11.061) SimpleApi: LightcapMode (T20) 1 -> 2 (4)
( 20.211) SimpleApi: 20.211197 T20 Device turned off: 2
( 20.212) SimpleApi: MPFIT stats for T20:
( 20.212) SimpleApi: 	seed runs         0 / 0
( 20.212) SimpleApi: 	error failures    0
( 20.249) SimpleApi: Adding tracked object T20 from HTC
```

jamuus, 2022-03-04:
Try this change see if it fixes the issue:

```
diff --git a/src/driver_vive.libusb.h b/src/driver_vive.libusb.h
index 9cefd40..2194bb1 100644
--- a/src/driver_vive.libusb.h
+++ b/src/driver_vive.libusb.h
@@ -232,10 +232,6 @@ static void handle_transfer(struct libusb_transfer *transfer) {
 
        iface->last_submit_time = OGGetAbsoluteTimeUS();
 
-       // If we get at least one packet; start applying a timeout
-       if (iface->assoc_obj && iface->assoc_obj->object_type != SURVIVE_OBJECT_TYPE_HMD) {
-               transfer->timeout = 1000;
-       }
        if (libusb_submit_transfer(transfer)) {
                goto shutdown;
        }
```

reyrael, 2022-03-04:
Thanks for the quick response. I will try it soon.

reyrael, 2022-03-04:
@jamuus I rebuilt it and i am still getting ```T20 Device turned off: 2```. I tried this with tracker 2 and tracker 3.

```
~/libsurvive $ ./bin/survive-cli
Info: Loaded drivers: GlobalSceneSolver, HTCVive
Info: Adding tracked object T20 from HTC
Info: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
Info: Detected LH gen 2 system.
Info: OOTX not set for LH in channel 0; attaching ootx decoder using device T20
Info: LightcapMode (T20) 1 -> 2 (4)
Warning: 1.826583 T20 Device turned off: 2
Info: MPFIT stats for T20:
Info: 	seed runs         0 / 0
Info: 	error failures    0
Info: Adding tracked object T20 from HTC
Info: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
Info: LightcapMode (T20) 1 -> 2 (4)
Warning: 10.168991 T20 Device turned off: 2
Info: MPFIT stats for T20:
Info: 	seed runs         0 / 0
Info: 	error failures    0
Info: Adding tracked object T20 from HTC
Info: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
Info: LightcapMode (T20) 1 -> 2 (4)
^CInfo: MPFIT stats for T20:
Info: 	seed runs         0 / 0
Info: 	error failures    0
```

Here it is with more logging:
```
~/libsurvive $ ./bin/survive-cli --v 100
Info: libsurvive version v1.01-4-g03190a5 (backend Eigen)
Info: Config file is /home/pi/.config/libsurvive/config.json
Info: Args: 
Info: 	'./bin/survive-cli'
Info: 	'--v'
Info: 	'100'
Info: Adding LH 1 mode: 0 id: 00000000
Info: Available Posers:
Info: 	 Dummy
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 KalmanOnly
Info: Totals 4 Posers.
Info: Using 'MPFIT' for poser
Info: Available Disambiguators:
Info: 	*StateBased
Info: Totals 1 Disambiguators.
Info: Using 'StateBased' for disambiguator
Info: libusb version 1.0.24.11584 (log level 2)
Info: Device added 0x5591c3a3b0
Info: Device added 0x5591c39e30
Info: Device added 0x5591c3a120
Info: Device added 0x5591c3b300
Info: Device added 0x5591c3aee0
Info: Enumerating USB device 28de:2300 Tracker (2018)
Info: Successfully enumerated Tracker (2018) (3) 28de:2300 at 0.0433748
Info: Requesting config for Tracker (2018) (nil) 0
Info: Device added 0x5591c58b80
Info: Device added 0x5591c58a90
Info: Loaded drivers: GlobalSceneSolver, HTCVive
Info: Available Posers:
Info: 	 Dummy
Info: 	 BaryCentricSVD
Info: 	*MPFIT
Info: 	 KalmanOnly
Info: Totals 4 Posers.
Info: Using 'MPFIT' for poser
Info: Creating threaded poser for T20
[... 131 lines cut ...]
Info: Submitting magic vive_magic_enable_lighthouse_more at 0.776680 sec for T20 - Tracker (2018) (length 5)
Info: Magics done in 0.777427 sec for T20 Tracker (2018)
Info: Device T20 has watchman FW version 1592875850 and FPGA version 538/7/2; named '                       watchman'. Hardware id 0x84020109 Board rev: 3 (len 56)
Info: Read config for T20
Info: Version done in 0.799843 sec for T20
Info: Setting T20 gyro scale 3(0.000244) acc scale 2(0.001065)
Info: Cleanup config for T20 Tracker (2018) at 2.694457 0/1
Info: Attaching IMU(0x81) for T20
Info: Attaching Lightcap(0x83) for T20
Info: Attaching Buttons(0x84) for T20
Info: Initializing general optimizer:
Info: 	max-error: 0.100000
Info: 	successes-to-reset: -1
Info: 	failures-to-reset: -1
Info: 	seed-poser: BaryCentricSVD
**Warning: 3.700190 T20 Device turned off: 2**
Info: Closing device on T20 (nil) (0x5591c4cd60)
Info: Cleaning up interface on 7 T20 IMU (0x5591c554d0)
Info: Cleaning up interface on 11 T20 Lightcap (0x5591c49b80)
Info: Cleaning up interface on 15 T20 Buttons (0x5591c49c70)
Info: Requesting close for unknown
Info: Closing device T20 Tracker (2018) (0/1)
Info: Disconnecting device T20 at 3.7011100

```