collabora/libsurvive #268: survive-cli on Windows returns "Failed to submit transfer unknown Watchan (-1)"

brianlah87, 2022-03-25:
I'm attempting to run survive-cli (1.0, from the Release Binaries) on Windows without a HMD. I have a Vive Tracker (2nd Gen) and the included dongle, and 2x 2nd Gen lighthouses.

I've confirmed that the dongle is paired using SteamVR, and confirmed that SteamVR is not running.

I get the following, and no further lines.

config.json is not automatically created as far as I can see.

I have not tried compiling from source - I figured I'd see if the binaries work first before diving any deeper.

Warning: Hotplug capabilities are not supported on this platform
Info: All enumerated devices attached.
Info: Loaded drivers: GlobalSceneSolver, HTCVive
Warning: Failed to submit transfer unknown Watchman (-1)

What am I missing? Is running it without a HMD attached a supported use case?

nagyidaiPeter, 2022-03-26:
> Update, I tried running the sample C# wrapper code and I'm getting:
>
>All enumerated devices attached.
>No USB devices detected
>Driver HTCVive reports status -1
>Loaded drivers: GlobalSceneSolver

Having the same issue, with the same devices.

trosacz, 2022-09-21:
have found any solution? same result. only works wired with USB cable dirrectly to tracker / controller.

heyangHEY, 2022-10-15:
Not exactly the same warning:
> Warning: Hotplug capabilities are not supported on this platform
> Info: All enumerated devices attached.                                                                                  
> Info: Loaded drivers: GlobalSceneSolver, HTCVive                                                                        
> Warning: Failed to submit transfer unknown Knuckles (-1)                                                                
> Warning: Failed to submit transfer unknown Knuckles (-1)                                                                
> Warning: Failed to submit transfer unknown Knuckles (-1)

KoenigSimon, 2023-02-13:
I have the same issue, no matter what dongle / wired / wireless

KoenigSimon, 2023-02-13:
It has been fixed in the later version 1.01 [here](https://github.com/cntools/libsurvive/releases/tag/v1.01)