collabora/libsurvive #277: mp_qrsolv: Assertion `isfinite(sdiag[i])' failed.

happysmash27, 2022-07-20:
**Describe the bug**
When trying to run libsurvive, it crashes with `survive-cli: /var/tmp/portage/xr-libs/libsurvive-1.01/work/libsurvive/redist/mpfit/mpfit.c:1656: mp_qrsolv: Assertion `isfinite(sdiag[i])' failed.`. 

**Log**
```
 % survive-cli
Info: Loaded drivers: GlobalSceneSolver, HTCVive
libusb: error [udev_hotplug_event] ignoring udev action change
Info: Adding tracked object HMD from HTC
Info: Device HMD has watchman FW version 1462663157 and FPGA version 262/0/0; named '                  steamservices'. Hardware id 0x80020100 Board rev: 128 (len 56)
Info: HMD is treated as HMD device
Info: Detected LH gen 1 system.
Info: Locked onto state 11(12,  1316733) at     68489398 for HMD
Info: OOTX not set for LH 0; attaching ootx decoder using device HMD
Info: OOTX not set for LH 1; attaching ootx decoder using device HMD
survive-cli: /var/tmp/portage/xr-libs/libsurvive-1.01/work/libsurvive/redist/mpfit/mpfit.c:1656: mp_qrsolv: Assertion `isfinite(sdiag[i])' failed.
zsh: IOT instruction  survive-cli
```

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 5.18.10-gentoo

**Additional context**
Have re-enabled udev rules (sometimes it seems like they may interfere with SteamVR so I have disabled an re-enabled them) and the same bug still occurs. SteamVR works without issue and after starting then closing it the logs are virtually identical. This bug never occurred before – my previous install before version 1.0 was rock solid – but now libsurvive always seems to fail with some kind of IOT instruction or another.

kwahoo2, 2023-07-15:
Hi,
Have you tried deleting 
`~/.config/libsurvive/config.json`?

This worked for me - Fedora 37.

throni3git, 2024-09-02:
i found a bug fix for MPFit at [https://pages.physics.wisc.edu/~craigm/idl/cmpfit.html](https://pages.physics.wisc.edu/~craigm/idl/cmpfit.html). the version used in libsurvive is that from 2016 i guess - and this was adapted to libsurvive regarding data types. in the two following patches of MPFit there were only few lines changed. one fix is already applied by the changes in libsurvive (mpfit.c -> mp_fdjac2 -> ij increment). but there is another fix that might solve the issue with `mp_qrsolv: Assertion `isfinite(sdiag[i])'`: in `mp_fdjac2` there is a calculation

`fjac[ij] = (fjac[ij] - wa[i]) / (2 * h); /* fjac[i+m*j] */`

which should be

`fjac[ij] = (wa2[ij] - wa[i]) / (2 * h); /* fjac[i+m*j] */`

according to the MPFit patch.