collabora/libsurvive #315: libsurvive Does not compile at this time on Fedora Kinoite 40

Jhdeval, 2024-10-14:
**Describe the bug**
The below error is shown during the make command. It appears to occur do to SciPlot having been installed. I attempted to see if simply changing the variable from plot to plot1 would fix the issue but I got the same thing. This lead me back to SciPlot but I am not sure where the include is being brought in from. C++ is not my regular language and I don't know the ins and outs of it.

/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:16:23: error: cannot declare field ‘cnkalman::ModelPlot::plot’ to be of abstract type ‘sciplot::Plot’
   16 |         sciplot::Plot plot;

**Compile log**
In file included from /var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc:1:
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:16:23: error: cannot declare field ‘cnkalman::ModelPlot::plot’ to be of abstract type ‘sciplot::Plot’
   16 |         sciplot::Plot plot;
      |                       ^~~~
In file included from /usr/include/sciplot/Plot2D.hpp:37,
                 from /usr/include/sciplot/Figure.hpp:34,
                 from /usr/include/sciplot/Canvas.hpp:33,
                 from /usr/include/sciplot/sciplot.hpp:38,
                 from /var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:6:
/usr/include/sciplot/Plot.hpp:56:7: note:   because the following virtual functions are pure within ‘sciplot::Plot’:
   56 | class Plot
      |       ^~~~
/usr/include/sciplot/Plot.hpp:204:18: note:     ‘virtual std::string sciplot::Plot::repr() const’
  204 |     virtual auto repr() const -> std::string = 0;
      |                  ^~~~
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:17:23: error: cannot declare field ‘cnkalman::ModelPlot::map’ to be of abstract type ‘sciplot::Plot’
   17 |         sciplot::Plot map;
      |                       ^~~
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc: In destructor ‘cnkalman::ModelPlot::~ModelPlot()’:
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc:66:18: error: ‘class sciplot::Plot’ has no member named ‘show’
   66 |             plot.show();
      |                  ^~~~
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc:67:17: error: ‘class sciplot::Plot’ has no member named ‘show’
   67 |             map.show();
      |                 ^~~~
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc:69:14: error: ‘class sciplot::Plot’ has no member named ‘save’
   69 |         plot.save(name + "-plot.svg");
      |              ^~~~
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc:70:13: error: ‘class sciplot::Plot’ has no member named ‘save’
   70 |         map.save(name + ".svg");
      |             ^~~~
/var/home/jhdeval/Nextcloud/Downloads/monado/libsurvive/libsurvive/libs/cnkalman/src/ModelPlot.cc:71:13: error: ‘class sciplot::Plot’ has no member named ‘save’
   71 |         map.save(name + ".png");
      |             ^~~~
[67/92] Building CXX object libs/cnkalman/libs/cnmatrix/src/CMakeFiles/cnmatrix.dir/eigen/svd.cpp.o
ninja: build stopped: subcommand failed.

Jhdeval, 2024-10-15:
It looks like 0.2 amd 0.3 of sciplot had breaking changes implemented. I went through the documentation in SAciPlot and was able to make the change to compile libsurvive. However, when I run survive-cli I get the following

/usr/local/bin/survive-cli: error while loading shared libraries: libsurvive.so.0: cannot open shared object file: No such file or directory

If I run survie-cli from the bin directory it appears works as expected. I believe this may be because Kinoite is an immutable OS and store the file in /usr/local instead of straight /usr. I am not sure how to get around this or if it is even possible.

ninimatteo, 2024-12-11:
I have the same issue on Ubuntu 20.04. If you found a solution please share it.
Thank you very much in advance!