collabora/libsurvive #322: Cannot build libsurvive (actually can't clone it)

astro-friedel, 2025-01-29:
**Describe the bug**
When following the instructions on the README, I get an error when trying to clone the repo.
```bash
$ git clone https://github.com/cntools/libsurvive.git --recursive

Cloning into 'libsurvive'...
remote: Enumerating objects: 14836, done.
remote: Counting objects: 100% (1078/1078), done.
remote: Compressing objects: 100% (75/75), done.
remote: Total 14836 (delta 1038), reused 1003 (delta 1003), pack-reused 13758 (from 2)
Receiving objects: 100% (14836/14836), 7.33 MiB | 5.68 MiB/s, done.
Resolving deltas: 100% (10884/10884), done.
Submodule 'libs/cnkalman' (https://github.com/collabora/cnkalman.git) registered for path 'libs/cnkalman'
Cloning into '/home/friedel/devel/ILLIXR-plugins/libsurvive/libs/cnkalman'...
remote: Repository not found.
fatal: repository 'https://github.com/collabora/cnkalman.git/' not found
fatal: clone of 'https://github.com/collabora/cnkalman.git' into submodule path '/home/friedel/devel/ILLIXR-plugins/libsurvive/libs/cnkalman' failed
Failed to clone 'libs/cnkalman'. Retry scheduled
Cloning into '/home/friedel/devel/ILLIXR-plugins/libsurvive/libs/cnkalman'...
remote: Repository not found.
fatal: repository 'https://github.com/collabora/cnkalman.git/' not found
fatal: clone of 'https://github.com/collabora/cnkalman.git' into submodule path '/home/friedel/devel/ILLIXR-plugins/libsurvive/libs/cnkalman' failed
Failed to clone 'libs/cnkalman' a second time, aborting

```
It appears that the repo https://github.com/collabora/cnkalman.git does not exist.

**Data**
None

**Compile log**
See above.

**Hardware setup**
Not relevant

**Desktop (please complete the following information):**
Ubuntu 24

Jebbly, 2025-01-29:
I believe this can be resolved by replacing the URL in ``.gitsubmodules`` with ``https://github.com/cntools/cnkalman``. Alternatively, the cnkalman repository could also be migrated to the collabora organization?

fredinfinite23, 2025-01-31:
We have transferred the two cntools projects below to collabora/org :
 - cntools/cnkalman
 - cntools/cnmatrix

and this issue is resolved. Please reclone

closed 2025-01-31 by fredinfinite23