collabora/libsurvive #323: Import or remove dependency on libsurvive-extras-data

davide125, 2025-02-17:
Currently libsurvive attempts to pull in [libsurvive-extras-data](https://github.com/jdavidberger/libsurvive-extras-data) as part of the [test suite](https://github.com/collabora/libsurvive/blob/4fb6d888d0277a8a3ba725e63707434d80ecdb2a/src/test_cases/CMakeLists.txt#L29-L31). This is a third party repo (apparently related to [libsurvive-extra](https://github.com/jdavidberger/libsurvive-extras)) that would probably make sense to pull into this org. Critically though, this repo currently lacks any [license attribution](https://github.com/jdavidberger/libsurvive-extras-data/issues/1), which makes it technically not redistributable.

jdavidberger, 2025-02-18:
I've added the license file; we can move the repo over to collabora too if that is useful

davide125, 2025-02-18:
Thanks @jdavidberger!

closed 2025-06-17 by bl4ckb0ne