collabora/libsurvive #324: low tracking quality

carlmstott, 2025-02-27:
Hello,

I'm writing a basic ros2 wrapper around libsurvive (all it does is scrapes the poses of the headsets/controllers and broadcasts them as transforms) and I'm getting very inconsistent tracking quality and I'm constantly hit with messages like:
Warning: Disambiguator got lost at 2819285738; refinding state for WM0
Info: Locked onto state  5( 6,   634218) at   2821685058 for WM0

or libsurvive crashing because the solver fails to converge 

these issues are accompanied with the pose's of the controllers/headset in my scene moving very erratically (zooming off into space in Rviz, etc). Ive also noticed that once this happens I need to delete the calibration data in order to get the controllers to behave again.

Is this a sign that my room setup is subpar? or should I be using a different solver (follow up question would be, how do I select the solver when I launch libsurvive)? Do you have any advice on how to tell if the light or reflective objects in my space could be contributing to my issues? I'm fairly sure its not how I'm writing my wrapper because I get the same issues with Andrew Symington's ros2 wrapper as well. 

any advice on bug squashing would be really appreciated.

ChristophHaag, 2025-03-05:
just making sure you have seen it, there is already some ros code in https://github.com/collabora/libsurvive/tree/master/tools/ros_publisher

Then I'm not sure if that's actually what your problem is but the global scene online calibration never worked well for me and was more likely to mess up a good calibration, so usually I'd

1. Delete the old calibration in ~/.config/libsurvive/config.json
2. run survive_cli and calibrate by putting the device(s) in a few different places
3. Run the actual app I want to run with the environment variable SURVIVE_GLOBALSCENESOLVER=0 (should also be possible to set in config.json somehow)

I feel like tracking has been better when I have 2 basestations in front of me, one left and one right, rather than on opposite sides of the room. But it could also be a placebo. Also tracking generally doesn't work great if the device is very close to a basestation (<20 cm).

Generally the tracking quality isn't great but should be "ok". Improvements always welcome.

carlmstott, 2025-03-05:
Thanks Christoph I appreciate your response. Unfortunately that code you linked to is written for ROS 1, and I'm using ROS 2, but I appreciate it anyway. 

Have you found more consistent behavior with the other posers besides MPFIT? I cant verify this because I cant specify launch arguments inside the wrapper I am writing (yet) but have you noticed any differences when using the different solvers?

carlmstott, 2025-03-06:
wow moving both of the base stations to the same side of the room has made a HUGE difference! thanks for the advice!!

throni3git, 2025-03-10:
in my experience it is crucial to avoid hard reflecting surfaces in the line-of-sight of the lighthouses. whenever i am in front of a window, this butterfly-like movement behaviour shows up like in your video. i also banned chromed furniture elements from my room. these really mess with the tracking.

do you have reflecting objects in your room that are now out of sight for the lighthouses?

agalakhov, 2025-03-10:
I experimented with tracking in my setup without any reflective surfaces. SteamVR works fine in my setup while libsurvive has tracking issues even with SteamVR calibration. My observations suggest that the Kalman filter is somehow parametrized so that it amplifies noise instead of reducing it. For example, ignoring IMU data seems to improve tracking and reduces sudden jerks. What comes from optical sensors looks perfect to me, but it gets worse while filtering.

My current observation is, setting IMU movement thresholds to zero improves things. I believe using thresholds is bad idea anyway since the filter then neglects significant data just below the threshold. But the tracking is still not good enough for battles in HL:Alyx while the game as such is playable.

agalakhov, 2025-03-11:
You may try to cover them or to look at them using any low-end camera. Low-end cameras usually see IR as blue so you can verify if you can see a reflection of the lighthouse. From my experience the reflective surface should be big like a wall mirror or a window in order to produce significant tracking issues. A picture on the wall high enough may cause problems too. Things on the desk are usually too low to cause issues at the normal height but a polished desk itself could be  problem. The lighthouse beam has to be systematically reflected in the direction of the headset, so anything that reflects in other directions is of no concern.

throni3git, 2026-08-27:
> I experimented with tracking in my setup without any reflective surfaces. SteamVR works fine in my setup while libsurvive has tracking issues even with SteamVR calibration. My observations suggest that the Kalman filter is somehow parametrized so that it amplifies noise instead of reducing it. For example, ignoring IMU data seems to improve tracking and reduces sudden jerks. What comes from optical sensors looks perfect to me, but it gets worse while filtering.
> 
> My current observation is, setting IMU movement thresholds to zero improves things. I believe using thresholds is bad idea anyway since the filter then neglects significant data just below the threshold. But the tracking is still not good enough for battles in HL:Alyx while the game as such is mostly playable.

how did you set the values then? i think you mean the config items in `src/survive_sensor_activations.c`?

agalakhov, 2026-08-27:
To be honest, I don't remember. I believe I edited libsurvive's configs.

ParkNoah821, 2026-09-26:
I looked through the IMU movement threshold and how it propagates through the tracking pipeline, and I have a hypothesis that might be worth testing.

As I understand the current code, the IMU movement threshold does not directly reject IMU measurements from the Kalman filter. Rather, it affects `last_movement`, which is then used to classify the object as stationary or moving. That classification changes the behavior of the poser, including the sensor time window, velocity modeling, and the use of the accelerometer-derived up-vector constraint.

This seems potentially relevant to the observation in this issue that reducing/ignoring IMU input can reduce sudden jerks, and that setting the IMU movement threshold to zero appeared to improve tracking.

I wonder if it would be worth separating two different concepts that are currently coupled through the movement threshold:

1. **How much we trust the IMU measurements**
2. **Whether the tracked object is physically stationary**

For the first one, instead of using a hard movement threshold to determine the influence of IMU data, perhaps the IMU measurements could always remain part of the filter, while increasing their measurement covariance `R` when the IMU is considered less reliable. This would reduce the Kalman gain and therefore reduce the influence of the IMU without effectively switching its contribution on/off.

For the second one, if the poser still needs a stationary/moving classification, perhaps it could be derived from the estimated state rather than directly from raw IMU changes. For example, stationary could require both:

```
||v|| < v_threshold
||ω|| < ω_threshold
```

for some minimum duration, with hysteresis between the stationary and moving thresholds to avoid rapid switching.

This would make the distinction more explicit:

```
raw IMU
    ↓
Kalman filter
    ↓
estimated v, ω  →  stationary/moving classification
    ↓
poser mode
```

while the IMU covariance independently determines how strongly the IMU affects the estimated state.

I am not suggesting that this is necessarily the root cause of the jerks, but it seems like a testable alternative explanation for the observations in this issue.

Has anyone already tried something along these lines — particularly increasing the IMU covariance while removing the movement threshold, and/or determining the stationary state from the estimated velocity and angular velocity rather than the raw IMU movement threshold?

If so, I would be very interested in the results.