collabora/libsurvive #330: Please update dependencies

scaronni, 2025-03-26:
Please update the code for sciplot and gattlib, it makes it difficult to package the project for distributions otherwise.

Current versions of sciplot and gattlib:

- sciplot 0.3.1
- gattlib 0.7.2

Thanks!

scaronni, 2025-03-26:
`libsurvive` currently compiles with `sciplot` 0.2.2 and `gattlib` 0.4.13.

bl4ckb0ne, 2025-03-26:
Are you encountering compilation issues with more recent versions? I see in the code that we don't require a version. What platform are you on, and could you give me your cmake generation command so I can reproduce locally please.

On master I can't even get the `sciplot` to build because of a missing `Findsciplot.cmake` file

scaronni, 2025-03-27:
`gattlib`:

```
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:22:9: error: unknown type name ‘gatt_connection_t’; did you mean ‘gatt_connect_cb_t’?
   22 |         gatt_connection_t *gconn;
      |         ^~~~~~~~~~~~~~~~~
      |         gatt_connect_cb_t
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:56:32: error: unknown type name ‘gatt_connection_t’; did you mean ‘gatt_connect_cb_t’?
   56 | static uint32_t read_char_byte(gatt_connection_t *gatt_connection, uuid_t *uuid) {
      |                                ^~~~~~~~~~~~~~~~~
      |                                gatt_connect_cb_t
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:74:37: error: unknown type name ‘gatt_connection_t’; did you mean ‘gatt_connect_cb_t’?
   74 | static void ble_connect_device_done(gatt_connection_t *gatt_connection, void *arg) {
      |                                     ^~~~~~~~~~~~~~~~~
      |                                     gatt_connect_cb_t
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c: In function ‘ble_connect_device’:
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:139:9: error: unknown type name ‘gatt_connection_t’; did you mean ‘gatt_connect_cb_t’?
  139 |         gatt_connection_t *gatt_connection;
      |         ^~~~~~~~~~~~~~~~~
      |         gatt_connect_cb_t
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:142:27: error: implicit declaration of function ‘gattlib_connect_async’; did you mean ‘gattlib_connect’? [-Wimplicit-function-declaration]
  142 |         gatt_connection = gattlib_connect_async(connection->driver->adapter, addr, 0, ble_connect_device_done, connection);
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           gattlib_connect
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:142:87: error: ‘ble_connect_device_done’ undeclared (first use in this function); did you mean ‘ble_connect_device’?
  142 |         gatt_connection = gattlib_connect_async(connection->driver->adapter, addr, 0, ble_connect_device_done, connection);
      |                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                       ble_connect_device
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:142:87: note: each undeclared identifier is reported only once for each function it appears in
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c: In function ‘gatt_thread’:
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:171:46: error: passing argument 2 of ‘gattlib_adapter_open’ from incompatible pointer type [-Wincompatible-pointer-types]
  171 |         int ret = gattlib_adapter_open(NULL, &driver->adapter);
      |                                              ^~~~~~~~~~~~~~~~
      |                                              |
      |                                              void **
In file included from /builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:4:
/usr/include/gattlib.h:253:72: note: expected ‘gattlib_adapter_t **’ {aka ‘struct _gattlib_adapter **’} but argument is of type ‘void **’
  253 | int gattlib_adapter_open(const char* adapter_name, gattlib_adapter_t** adapter);
      |                                                    ~~~~~~~~~~~~~~~~~~~~^~~~~~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:178:60: error: passing argument 2 of ‘gattlib_adapter_scan_enable’ from incompatible pointer type [-Wincompatible-pointer-types]
  178 |         ret = gattlib_adapter_scan_enable(driver->adapter, ble_discovered_device, 1, driver);
      |                                                            ^~~~~~~~~~~~~~~~~~~~~
      |                                                            |
      |                                                            void (*)(void *, const char *, const char *, void *)
/usr/include/gattlib.h:274:89: note: expected ‘gattlib_discovered_device_t’ {aka ‘void (*)(struct _gattlib_adapter *, const char *, const char *, void *)’} but argument is of type ‘void (*)(void *, const char *, const char *, void *)’
  274 | int gattlib_adapter_scan_enable(gattlib_adapter_t* adapter, gattlib_discovered_device_t discovered_device_cb, size_t timeout, void *user_data);
      |                                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
[... cut ...]
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/src/driver_gatt.c:201:41: error: too few arguments to function ‘gattlib_disconnect’; expected 2, have 1
  201 |                                         gattlib_disconnect(driver->lhs[i]->gconn);
      |                                         ^~~~~~~~~~~~~~~~~~
/usr/include/gattlib.h:385:5: note: declared here
  385 | int gattlib_disconnect(gattlib_connection_t* connection, bool wait_disconnection);
      |     ^~~~~~~~~~~~~~~~~~
```

scaronni, 2025-03-27:
Relevant commit for `gattlib`: https://github.com/labapart/gattlib/commit/ce52533f395cdfbec024778a068366b4db035147

scaronni, 2025-03-27:
`sciplot`, error is in `cnkalman`:

```
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:16:23: error: cannot declare field ‘cnkalman::ModelPlot::plot’ to be of abstract type ‘sciplot::Plot’
   16 |         sciplot::Plot plot;
      |                       ^~~~
In file included from /usr/include/sciplot/Plot2D.hpp:37,
                 from /usr/include/sciplot/Figure.hpp:34,
                 from /usr/include/sciplot/Canvas.hpp:33,
                 from /usr/include/sciplot/sciplot.hpp:38,
                 from /builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:6:
/usr/include/sciplot/Plot.hpp:56:7: note:   because the following virtual functions are pure within ‘sciplot::Plot’:
   56 | class Plot
      |       ^~~~
/usr/include/sciplot/Plot.hpp:204:18: note:     ‘virtual std::string sciplot::Plot::repr() const’
  204 |     virtual auto repr() const -> std::string = 0;
      |                  ^~~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/../include/cnkalman/ModelPlot.h:17:23: error: cannot declare field ‘cnkalman::ModelPlot::map’ to be of abstract type ‘sciplot::Plot’
   17 |         sciplot::Plot map;
      |                       ^~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/ModelPlot.cc: In destructor ‘cnkalman::ModelPlot::~ModelPlot()’:
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/ModelPlot.cc:66:18: error: ‘class sciplot::Plot’ has no member named ‘show’
   66 |             plot.show();
      |                  ^~~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/ModelPlot.cc:67:17: error: ‘class sciplot::Plot’ has no member named ‘show’
   67 |             map.show();
      |                 ^~~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/ModelPlot.cc:69:14: error: ‘class sciplot::Plot’ has no member named ‘save’
   69 |         plot.save(name + "-plot.svg");
      |              ^~~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/ModelPlot.cc:70:13: error: ‘class sciplot::Plot’ has no member named ‘save’
   70 |         map.save(name + ".svg");
      |             ^~~~
/builddir/build/BUILD/libsurvive-1.01_20250326git8d4765a-build/libsurvive-8d4765a067ad2170f389a3d6226fe8bc555feb13/libs/cnkalman/src/ModelPlot.cc:71:13: error: ‘class sciplot::Plot’ has no member named ‘save’
   71 |         map.save(name + ".png");
      |             ^~~~
```

scaronni, 2025-03-27:
Here I'm not sure exactly where the changes are, as the `sciplot`  0.2.2 version required by `libsurvive` it's almost 4 years old.

scaronni, 2025-03-27:
btw, thanks for the latest commits about importing `cnkalman`, `cnmatrix` and dropping the python bindings, this has helped a lot in the packaging. Trying to compile all of it before as shared libraries has been quite a challenge.

bl4ckb0ne, 2025-03-27:
No idea why I can't reproduce on debian Trixie, but I booted an archlinux container and can see what you are reporting. I'll prepare a patch.

bl4ckb0ne, 2025-03-27:
I opened https://github.com/collabora/libsurvive/pull/331, took care of sciplot (0.3.1) already, i'll do gattlib later.

closed 2025-03-28 by bl4ckb0ne