collabora/libsurvive #334: Tundra tracker is not detected

Borshmann, 2025-05-19:
There are 2 HTC LightHouse and Tundra Labs trackers.
After compiling and running the survive-cli in the console, I see successful LH detection and their channels. 
But as far as I understand, both communication and the visibility of trackers don't work. I don't understand something and I'm obviously doing something wrong. 
I get a warning - "The detected version for device KN1 is 1637337510; the latest that is verified to work is 1632556731. You may have to upgrade libsurvive to support this device."
Help!

Info: Adding tracked object KN0 from HTC
Info: Adding tracked object KN1 from HTC
Info: Adding tracked object KN2 from HTC
Info: All enumerated devices attached.
Info: Loaded drivers: GlobalSceneSolver, HTCVive
Info: Device KN1 has watchman FW version 1637337510 and FPGA version 538/7/2; named '                   WMBUILD-W64$'. Hardware id 0xf1040009 Board rev: 3 (len 65)
Warning: The detected version for device KN1 is 1637337510; the latest that is verified to work is 1632556731. You may have to upgrade libsurvive to support this device.
Info: Detected LH gen 2 system.
Info: LightcapMode (KN1) 1 -> 2 (ff)
Info: OOTX not set for LH in channel 6; attaching ootx decoder using device KN1
Info: OOTX not set for LH in channel 1; attaching ootx decoder using device KN1
Info: MPFIT success 184657.877499/742.7446999278/0.0007388 (17 measurements, 1, MP_OK_CHI, 101 iters, up err 0.0000031, trace 0.0000156)
Info: Global solve with 1 scenes for 2 with error of 184657.877499/742.7446999278 (acc err 0.0016)
Info: Using LH 2 (87dc66e2) as reference lighthouse
Info: MPFIT success 3458094.384855/9158.7476456348/0.0013298 (57 measurements, 1, MP_OK_CHI, 41 iters, up err 0.0000355, trace 0.0001016)
Info: Global solve with 2 scenes for 1 with error of 3458094.384855/9158.7476456348 (acc err 0.0058)
Info: Global solve with 2 scenes for 2 with error of 3458094.384855/9158.7476456348 (acc err 0.0009)
Info: Using LH 2 (87dc66e2) as reference lighthouse
Info: MPFIT success 1222451.363078/18490.9931868173/0.0014941 (89 measurements, 1, MP_OK_CHI, 70 iters, up err 0.0001586, trace 0.0001488)
Info: Global solve with 3 scenes for 1 with error of 1222451.363078/18490.9931868173 (acc err 0.0045)
Info: Global solve with 3 scenes for 2 with error of 1222451.363078/18490.9931868173 (acc err 0.0012)
Info: Using LH 2 (87dc66e2) as reference lighthouse
Info: MPFIT success 2256131.979920/27215.1669179496/0.0015388 (122 measurements, 1, MP_OK_CHI, 6 iters, up err 0.0002119, trace 0.0001887)
Info: Global solve with 4 scenes for 1 with error of 2256131.979920/27215.1669179496 (acc err 0.0056)
Info: Global solve with 4 scenes for 2 with error of 2256131.979920/27215.1669179496 (acc err 0.0014)
Info: Using LH 2 (87dc66e2) as reference lighthouse

As far as I understand, he's trying to calibrate himself, but it doesn't end with anything. 
If you use sensors-readout, it does not see the tracker wirelessly, but reads it correctly over the wire.

Windows 11 23H2

bl4ckb0ne, 2025-05-20:
> I get a warning - "The detected version for device KN1 is 1637337510; the latest that is verified to work is 1632556731. You may have to upgrade libsurvive to support this device."
> But as far as I understand, there is no newer version and I don't understand how to solve this.

We probably need to update for the new version in the code. It looks like something change in the new firmware version but unfortunately i don't have the hw to work on that.

Borshmann, 2025-05-21:
Thanks! 
I tried all the release versions and on version 0.3 (which is amazing) My positioning is messed up. But it does not work stably. Sometimes, after turning off the devices and turning them on again, the devices do not understand each other and I get a set of random bytes in the output. This is (seemingly) treated by reconnecting trackers to Steam VR.

Do I understand correctly that the library does not have hw support for tundra tracker? 
If I take the trackers from HTC, will there be any problems? 

Thanks again for the reply.

Borshmann, 2025-05-21:
Thanks! 
I tried all release versions and on version 0.3 (surprisingly) I got the positioning working. But it is not stable. Sometimes, after turning off devices and turning them back on, the devices don't understand each other and I get a set of random bytes in the output. It is cured (sort of) by re-binding trackers in Steam VR.

Am I right that the library does not support hw for tundra tracker? 
If I take trackers from HTC - no problems? 

Thanks again for the answer.

Translated with DeepL.com (free version)

Borshmann, 2025-05-21:
Thanks! 
I tried all release versions and on version 0.3 (surprisingly) I got the positioning working. But it is not stable. Sometimes, after turning off devices and turning them back on, the devices don't understand each other and I get a set of random bytes in the output. It is cured (sort of) by re-binding trackers in Steam VR.

Am I right that the library does not support hw for tundra tracker? 
If I take trackers from HTC - no problems? 

Thanks again for the answer.

bl4ckb0ne, 2025-05-26:
> Am I right that the library does not support hw for tundra tracker?

Seems like it, sorry

kr15h, 2025-10-23:
I actually got Tundra Tracker working a year ago. Now I am trying it again. Will let you know if that works. As for this line `The detected version for device KN1 is 1637337510; the latest that is verified to work is 1632556731. You may have to upgrade libsurvive to support this device.`, it is just a number that can be changed in a file and does not affect functionality. Yes, the tracking is not as stable as SteamVR, but it works.