collabora/libsurvive #336: Prevent libsurvive from continously optimizing the calibration.

Tuntenfisch, 2025-07-04:
Hello,

is there an option from preventing libsurvive from changing the calibration file continuously. I want to deploy libsurvive tracking on multiple robots individually but noticed that the library keeps "optimizing" the individual lighthouse poses ever so slightly. This naturally leads to the coordinate systems of each robot shifting over time.

I already attempted to disable calibration with `--disable-calibrate` or setting `"PositionSet": "1"` for each lighthouse inside the calibration JSON (even though I have no clue what `PositionSet` does. I also tried just making the calibration file read-only but that doesn't stop libsurvive from loading the contents and simply changing the calibration in memory.

ChristophHaag, 2025-07-04:
Does the environment variable SURVIVE_GLOBALSCENESOLVER=0 do what you want?

Tuntenfisch, 2025-07-04:
I was also unable to find any reference in the code to `SURVIVE_GLOBALSCENESOLVER`, so I'm unsure where this mechanism is implemented.

Tuntenfisch, 2025-07-04:
Yeah, I just checked and the changes in position are quite large (24cm in one instance):

```bash
diff --git a/config/lighthouse_calibration.json b/config/lighthouse_calibration.json
index fbd1660..ccfdf66 100644
--- a/config/lighthouse_calibration.json
+++ b/config/lighthouse_calibration.json
@@ -8,8 +8,8 @@
 "index":"0",
 "id":"260523886",
 "mode":"1",
-"pose":["0.000493459520","5.398861885071","2.215373277664","0.120629101992","0.110769331455","-0.572545826435","-0.803349256516"],
-"variance":["0.000000000000","0.000000000000","0.000000000000","0.000000000000","0.000000000000","0.000000000000"],
+"pose":["0.240459725261","5.423994541168","2.125160694122","0.122166467632","0.084696790637","-0.582821001710","-0.798887656562"],
+"variance":["0.000004530067","0.000002909254","0.000005936905","0.005678600166","0.001677540597","0.003943505231"],
 "unlock_count":"0",
 "accel":["1.000000000000","127.000000000000","41.000000000000"],
 "fcalphase":["0.000000000000","-0.003702163696"],
@@ -20,7 +20,7 @@
 "fcalogeephase":["2.201171875000","2.968750000000"],
 "fcalogeemag":["0.187500000000","0.155395507812"],
 "OOTXSet":"1",
-"PositionSet":"0"
+"PositionSet":"1"
 }
 "lighthouse1":{
 "index":"1",
@@ -62,7 +62,7 @@
 "index":"3",
 "id":"2011527430",
 "mode":"0",
-"pose":["3.982059240341","1.159700155258","2.235741376877","-0.501658856869","-0.339023172855","-0.447369277477","-0.658226728439"],
+"pose":["4.086276531219","1.113584280014","2.086863994598","-0.499750854795","-0.353418739213","-0.458857121113","-0.644045355785"],
 "variance":["0.000000000000","0.000000000000","0.000000000000","0.000000000000","0.000000000000","0.000000000000"],
 "unlock_count":"0",
 "accel":["6.000000000000","127.000000000000","45.000000000000"],
@@ -74,5 +74,5 @@
 "fcalogeephase":["1.610351562500","2.332031250000"],
 "fcalogeemag":["-0.025344848633","-0.156372070312"],
 "OOTXSet":"1",
-"PositionSet":"0"
+"PositionSet":"1"
 }
```

Tuntenfisch, 2025-07-04:
Okay, when I manually change the calibration file, and for each lighthouse set

- `"variance":["0.000000000000","0.000000000000","0.000000000000","0.000000000000","0.000000000000","0.000000000000"],`
- `"PositionSet":"1"`

the calibration doesn't seem to be altered anymore. I suppose this is because the lighthouse poses are also continuously optimized  via some Kalman-Filter and a variance of `0` mathematically boils down to `new_lighthouse_pose = old_lighthouse_pose` for any pose update that might occur!? While my short testing showed that the calibration remains unchanged, I'm not very confident in it not changing at some point. Perhaps someone with deeper insight into the codebase can confirm my rationale?